project-chip/connectedhomeip #5937: Use chip-device-ctrl by BLE Connect to the device failed！

luyou-tuya, 2021-04-10:
#### 
Connect the device via chip-device-ctrl Bluetooth on the _**Raspberry Pi**_
Executing chip-device-ctrl shows the following _error：
/python# chip-device-ctrl
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x000000000001b669
CHIP:DL: CHIP task running
CHIP:DL: BLEManager initialization failed: Ble Error 6001 (0x00001771): Incorrect state
CHIP:DL: Failed to initialize CHIP stack: platform init failed: Ble Error 6001 (0x00001771): Incorrect state
CHIP:DL: Platform main loop started.
Chip Device Controller Shell

Then execute "connect -ble 3840 73141520 12344321“ The connection failed, the log shows as follows：
chip-device-ctrl > connect -ble 3840 73141520 12344321
Device is assigned with nodeid = 12344321
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:DL: TRACE: Bus acquired for name C-3fe6
CHIP:DL: PlatformBlueZInit init success
CHIP:BLE: BLE scanning through known devices.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: New device scanned: 68:68:0F:0E:CF:2A
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Scan complete notification without an active scan.
CHIP:DL: ConnectDevice complete
CHIP:DL: Char1 /org/bluez/hci0/dev_68_68_0F_0E_CF_2A/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_68_68_0F_0E_CF_2A/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_68_68_0F_0E_CF_2A/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_68_68_0F_0E_CF_2A/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_68_68_0F_0E_CF_2A/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_68_68_0F_0E_CF_2A/service0018
CHIP:DL: New BLE connection 0xffff9c043110, device 68:68:0F:0E:CF:2A, path /org/bluez/hci0/dev_68_68_0F_0E_CF_2A
CHIP:DL: HandlePlatformSpecificBLEEvent 16385
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: HandlePlatformSpecificBLEEvent 16388
CHIP:BLE: subscribe complete, ep = 0xffffa65cf188
CHIP:DL: Indication received, conn = 0xffff9c043110
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:BLE: peripheral chose BTP version 3; central expected between 2 and 3
CHIP:BLE: using BTP fragment sizes rx 20 / tx 128.
CHIP:BLE: local and remote recv window size = 3
CHIP:BLE: Assigned local session key ID 0
CHIP:BLE: Sent PBKDF param request
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff9c043110
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: Indication received, conn = 0xffff9c043110
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff9c043110
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: Indication received, conn = 0xffff9c043110
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff9c043110
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:BLE: Received PBKDF param response
CHIP:BLE: Sent spake2p msg1
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff9c043110
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff9c043110
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: Indication received, conn = 0xffff9c043110
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff9c043110
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: Indication received, conn = 0xffff9c043110
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[... 290 lines cut ...]
00> <detail> [DL]         FE80::B4F3:5E73:3DB3:8F00/64 valid preferred
00> <detail> [DL] LwIP Thread interface addresses updated
00> <detail> [DL]    FE80::B4F3:5E73:3DB3:8F00 IPv6 link-local address, preferred)
00> <detail> [DL]    FDDE:AD00:BEEF:0:12FF:A5E0:DB10:7E06 Thread mesh-local address, preferred)
00> <info  > [DL] _OnPlatformEvent default:  event->Type = 32778
00> <detail> [DL] OpenThread State Changed (Flags: 0x301b32b7)
00> <detail> [DL]    Device Role: CHILD
00> <detail> [DL]    Network Name: OpenThread-ba18
00> <detail> [DL]    PAN Id: 0xDEAD
00> <detail> [DL]    Extended PAN Id: 0x1111111133333333
00> <detail> [DL]    Channel: 15
00> <detail> [DL]    Mesh Prefix: FD76:BDD5:A365:4284::/64
00> <detail> [DL]    Partition Id: 0x52B0CF52
00> <detail> [DL]    Thread Unicast Addresses:
00> <detail> [DL]         FD76:BDD5:A365:4284:0:FF:FE00:A802/64 valid preferred rloc
00> <detail> [DL]         FD76:BDD5:A365:4284:12FF:A5E0:DB10:7E06/64 valid preferred
00> <detail> [DL]         FE80::B4F3:5E73:3DB3:8F00/64 valid preferred
00> <detail> [DL] LwIP Thread interface addresses updated
00> <detail> [DL]    FE80::B4F3:5E73:3DB3:8F00 IPv6 link-local address, preferred)
00> <detail> [DL]    FD76:BDD5:A365:4284:12FF:A5E0:DB10:7E06 Thread mesh-local address, preferred)

 #### Proposed Solution
Why this error occurs？

<suggested fix, suggested enhancement>

luyou-tuya, 2021-04-12:
@Damian-Nordic Please help me to solve this problem. Why does the connection fail and how to solve it .thank you very much!

Damian-Nordic, 2021-04-12:
@luyou-tuya It looks OK on the Python controller side - it's able to establish a secure session and send Thread credentials to the device. The device, however, is unable to join the Thread network and the rendenzvous session probably times out. Did you have border router running and Thread network formed when you ran the command?

Anyway, you seem to use SiLabs platform which I'm not familiar with enough. @jmartinez-silabs can you help @luyou-tuya get the Thread provisioning working?

luyou-tuya, 2021-04-13:
@Damian-Nordic @jmartinez-silabs  otbr has been running, thread network has been established. otbr and chip-device-ctrl work together on the Raspberry Pi
**The configuration of otbr is as follows:**
/Desktop# ot-ctl dataset
Active Timestamp: 1
Channel: 15
Channel Mask: 0x07fff800
Ext PAN ID: 1111111133333333
Mesh Local Prefix: fd59:75e8:f33:ec8::/64
Master Key: 00112233445566778899aabbccddeeff
Network Name: OpenThread-3881
PAN ID: 0xdead
PSKc: dda59e8255ebd78cbe8b6831f1cd9a57
Security Policy: 672, onrcb
Done
**the chip-efr32-lighting-example log  as follows:**
 OpenThread State Changed (Flags: 0x301b32b7)
00> <detail> [DL]    Device Role: CHILD
00> <detail> [DL]    Network Name: OpenThread-3881
00> <detail> [DL]    PAN Id: 0xDEAD
00> <detail> [DL]    Extended PAN Id: 0x1111111133333333
00> <detail> [DL]    Channel: 15
00> <detail> [DL]    Mesh Prefix: FD59:75E8:F33:EC8::/64
00> <detail> [DL]    Partition Id: 0x69FE97C2
00> <detail> [DL]    Thread Unicast Addresses:
00> <detail> [DL]         FD59:75E8:F33:EC8:0:FF:FE00:B803/64 valid preferred rloc
00> <detail> [DL]         FD59:75E8:F33:EC8:5864:817A:BDAB:A637/64 valid preferred
00> <detail> [DL]         FE80::1049:EA8B:102A:2F00/64 valid preferred
00> <detail> [DL] LwIP Thread interface addresses updated
00> <detail> [DL]    FE80::1049:EA8B:102A:2F00 IPv6 link-local address, preferred)
00> <detail> [DL]    FD59:75E8:F33:EC8:5864:817A:BDAB:A637 Thread mesh-local address, preferred)
00> <info  > [DL] Char Write Req, char : 26
**Does the log show that it has successfully joined the thread network?**

jmartinez-silabs, 2021-04-13:
From what I see, the secure rendez-vous completes successfuly and the controller provides the thread credentials. The device connects to the thread network but never gets a gobal link address so it doesn't send it back to the controller whos is waiting for it. He eventually times out and disconnect.

How di you form your thread network on your OTBR did you use the web gui? If you did it manually did you provide a prefix like this `prefix add fd11:22::/64 paros med `

luyou-tuya, 2021-04-13:
@jmartinez-silabs yes! I tried like this！I execute the command like this 
**sudo ot-ctl prefix add 2001:db8::/64 paros 
But the connection still fails** 
I think it is because the mdns function is not implemented that causes the connection to fail.
I use this example "examples/lighting-app/efr32"
see this compiled file src/platform/device.gni,  chi_mdns="none", Advertiser_ImplNone.cpp source file is compiled 
 # Select DNS-SD implementation
  if (chip_device_platform == "linux" || chip_device_platform == "esp32") {
    chip_mdns = "minimal"
  } else if (chip_device_platform == "darwin") {
    chip_mdns = "platform"
  } else {
    chip_mdns = "none"
  }
}
class NoneAdvertiser : public ServiceAdvertiser
{
public:
    CHIP_ERROR Start(chip::Inet::InetLayer * inetLayet, uint16_t port) override
    {
        **ChipLogError(Discovery, "mDNS advertising not available. mDNS start disabled.");**
        return CHIP_ERROR_NOT_IMPLEMENTED;
    }
}
**_I think it may be that the mdns function is not implemented and the connection fails!_**.Do you think this may be the reason?

jmartinez-silabs, 2021-04-13:
I think it should work without the DNS-SD implementation still. It definetly was last week. I can try on my setup here when i get a few minutes and let you know. 

However, I think you should try to stop and restart your thread network making sure your dataset is right before re-starting it. From the console logs you your device did not received any 2001:db8::/64 address and I really think this is the issue. You only received for your BR a link-local and mesh-local address

```
Thread Unicast Addresses:
00> [DL] FD59:75E8:F33:EC8:0:FF:FE00:B803/64 valid preferred rloc
00> [DL] FD59:75E8:F33:EC8:5864:817A:BDAB:A637/64 valid preferred
00> [DL] FE80::1049:EA8B:102A:2F00/64 valid preferred
00> [DL] LwIP Thread interface addresses updated
00> [DL] FE80::1049:EA8B:102A:2F00 IPv6 link-local address, preferred)
00> [DL] FD59:75E8:F33:EC8:5864:817A:BDAB:A637 Thread mesh-local address, preferred)
```

luyou-tuya, 2021-04-13:
@jmartinez-silabs  ok! I am trying to do this. This log is that I did not add prefix on the border route，It’s too late today, my environment is destroyed. I’ll try it tomorrow and I’ll tell you ，thanks for your help！

jmartinez-silabs, 2021-04-13:
@luyou-tuya I was able to complete the connection and provisionning without the srp enable on current master

.....
CHIP:IN: New pairing for device 0x0000000000bc5c01, key 0!!
CHIP:CTL: Remote device completed SPAKE2+ handshake

CHIP:NP: Sending Thread Credentials
CHIP:IN: Secure message was encrypted: Msg ID 0
CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000bc5c01 at utc time: 20817395 msec
CHIP:IN: Sending secure msg on connection specific transport
CHIP:IN: Secure msg send status No Error
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff9403f200
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff9403f200
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff9403f200
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: Indication received, conn = 0xffff9403f200
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff9403f200
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:NP: Received kIPAddressAssigned
CHIP:CTL: Remote device was assigned an ip address

CHIP:CTL: AsyncSetKeyValue: PairedDevicebc5c01=EAC+PAg3eO30SlFfHuyPtucXAAAAAAAAAAAAAAAAAAAAAAEAAAAAAAFcvAAAAAAAZmQxMToyMjo6ZDZkMDo5NzFlOjIzM2Y6YzlkMAAAAAAAAAAAAAAAAAAAAAAAAFkrAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
CHIP:CTL: AsyncSetKeyValue: ListPairedDevices0=AVy8AAAAAAA=
CHIP:CTL: AsyncSetKeyValue: StartKeyID=1
Secure Session to Device Established
Device temporary node id (**this does not match spec**): 12344321
chip-device-ctrl > CHIP:DL: HandlePlatformSpecificBLEEvent 16388
CHIP:BLE: Auto-closing end point's BLE connection.
CHIP:DL: Closing BLE GATT connection (con 0xffff9403f200)
CHIP:DL: BluezDisconnect peer=71:A3:14:08:EE:A5
CHIP:DL: Bluez disconnected
CHIP:DL: Bluez notify CHIPoBluez connection disconnected
chip-device-ctrl >

luyou-tuya, 2021-04-14:
@jmartinez-silabs Thank you for your help ! The reason is that this command ot-ctl prefix add fd11:22::/64 paros did not take effect after execution
The correct otbr configuration script is as follows, I hope it will be helpful to others：
ot-ctl factoryreset
sleep 1
ot-ctl dataset init new
sleep 1
ot-ctl dataset channel 15
sleep 1
ot-ctl dataset panid 0x1234
sleep 1
ot-ctl dataset extpanid  1111111122222222
sleep 1
ot-ctl dataset masterkey 00112233445566778899aabbccddeeff
sleep 1
ot-ctl prefix add 2001:db8::/64 paros
sleep 1
ot-ctl dataset commit active
sleep 1
ot-ctl ifconfig up
sleep 1
ot-ctl thread start

closed 2021-04-14 by luyou-tuya