project-chip/connectedhomeip #5961: Crash in -[BleConnection start]

terafin, 2021-04-12:
#### Problem
      7 libdispatch.dylib: _dispatch_lane_resume
 ==> 7 CHIP: -[BleConnection start] <==
          7 CHIP: -[BleConnection centralManagerDidUpdateState:]
            7 CoreBluetooth: -[CBCentralManager observeValueForKeyPath:ofObject:change:context:]
              7 Foundation: NSKeyValueNotifyObserver
                7 Foundation: NSKeyValueDidChange
                  7 Foundation: -[NSObject(NSKeyValueObservingPrivate) _changeValueForKeys:count:maybeOldValuesDict:maybeNewValuesDict:usingBlock:]
                    7 Foundation: -[NSObject(NSKeyValueObservingPrivate) _changeValueForKey:key:key:usingBlock:]
                      7 Foundation: _NSSetLongLongValueAndNotify
                        7 CoreBluetooth: -[CBManager extractLocalDeviceStatesDictionary:]
                          7 CoreBluetooth: -[CBManager xpcConnectionDidReceiveMsg:args:]
                            7 CoreBluetooth: __30-[CBXpcConnection _handleMsg:]_block_invoke
                              7 libdispatch.dylib: _dispatch_call_block_and_release
                                7 libdispatch.dylib: _dispatch_client_callout
                                  7 libdispatch.dylib: _dispatch_lane_serial_drain
                                    7 libdispatch.dylib: _dispatch_lane_invoke
                                      7 libdispatch.dylib: _dispatch_lane_serial_drain
                                        7 libdispatch.dylib: _dispatch_lane_invoke
                                          7 libdispatch.dylib: _dispatch_workloop_worker_thread
                                            7 libsystem_pthread.dylib: _pthread_wqthread
                                              7 libsystem_pthread.dylib: 

 #### Proposed Solution
Find/fix :)

woody-apple, 2021-04-14:
@vivien-apple could you take a peek?

vivien-apple, 2021-04-15:
@terafin Can you give me more contexts about how to reproduce this issue ?
For example, was is using `chip-tool` or `iOS ChipTool` ?
Also, can you reproduce it reliably, or has it happens only once ?

vivien-apple, 2021-04-20:
\o/ 
I found a way to reproduce the same stack by turning the bluetooth on/off during the pairing process.

PR #6169 (2021-04-21): Crash in -[BleConnection start]
 #### Problem
Problem

  7 libdispatch.dylib: _dispatch_lane_resume
==> 7 CHIP: -[BleConnection start] <==
7 CHIP: -[BleConnection centralManagerDidUpdateState:]
7 CoreBluetooth: -[CBCentralManager observeValueForKeyPath:ofObject:change:context:]
7 Foundation: NSKeyValueNotifyObserver
7 Foundation: NSKeyValueDidChange
7 Foundation: -[NSObject(NSKeyValueObservingPrivate) _changeValueForKeys:count:maybeOldValuesDict:maybeNewValuesDict:usingBlock:]
7 Foundation: -[NSObject(NSKeyValueObservingPrivate) _changeValueForKey🔑key:usingBlock:]
7 Foundation: _NSSetLongLongV [... cut]
closed 2021-04-21 by woody-apple