project-chip/connectedhomeip #5962: Crash in chip::System::Object::SystemLayer

woody-apple, 2021-04-12:
#### Problem
Crash in CHIP:

```
Crashpoint: CHIP:  chip::System::Object::SystemLayer const:163
Exception: EXC_BAD_ACCESS (SIGSEGV)

 Thread 9 Crashed:
 0   CHIP                          	 0x00000001ae4be618 chip::System::Object::SystemLayer() const + 12 (SystemObject.h:144)
 1   CHIP                          	 0x00000001ae483508 chip::Inet::UDPEndPoint::Listen() + 36 (UDPEndPoint.cpp:312)
 2   CHIP                          	 0x00000001ae517cd0 chip::Transport::UDP::Init(chip::Transport::UdpListenParameters&amp;) + 268 (UDP.cpp:55)
 3   CHIP                          	 0x00000001ae474f28 int chip::Transport::Tuple<chip::Transport::UDP, chip::Transport::UDP>::InitImpl<chip::Transport::UdpListenParameters&amp;>(chip::Transport::RawTransportDelegate*, chip::Transport::UdpListenParameters&amp;&amp;&amp;) + 76 (Tuple.h:203)
 4   CHIP                          	 0x00000001ae474ea8 int chip::Transport::Tuple<chip::Transport::UDP, chip::Transport::UDP>::InitImpl<chip::Transport::UdpListenParameters&amp;, chip::Transport::UdpListenParameters&amp;>(chip::Transport::RawTransportDelegate*, chip::Transport::UdpListenParameters&amp;&amp;&amp;, chip::Transport::UdpListenParameters&amp;&amp;&amp;) + 148 (Tuple.h:211)
 5   CHIP                          	 0x00000001ae474df4 int chip::Transport::Tuple<chip::Transport::UDP, chip::Transport::UDP>::Init<chip::Transport::UdpListenParameters&amp;, chip::Transport::UdpListenParameters&amp;, (void*)0>(chip::Transport::RawTransportDelegate*, chip::Transport::UdpListenParameters&amp;&amp;&amp;, chip::Transport::UdpListenParameters&amp;&amp;&amp;) + 88 (Tuple.h:106)
 6   CHIP                          	 0x00000001ae46f28c int chip::TransportMgr<chip::Transport::UDP, chip::Transport::UDP>::ResetTransport<chip::Transport::UdpListenParameters&amp;, chip::Transport::UdpListenParameters&amp;>(chip::Transport::UdpListenParameters&amp;&amp;&amp;, chip::Transport::UdpListenParameters&amp;&amp;&amp;) + 84 (TransportMgr.h:77)
 7   CHIP                          	 0x00000001ae46e480 chip::Controller::Device::LoadSecureSessionParameters(chip::Controller::Device::ResetTransport) + 264 (CHIPDevice.cpp:325)
 8   CHIP                          	 0x00000001ae46e164 chip::Controller::Device::SendMessage(chip::System::PacketBufferHandle, chip::PayloadHeader&amp;) + 380 (CHIPDevice.cpp:89)
 9   CHIP                          	 0x00000001ae46e65c chip::Controller::Device::SendMessage(chip::System::PacketBufferHandle) + 84 (CHIPDevice.cpp:138)
 10  CHIP                          	 0x00000001ae4636b0 chip::Controller::ClusterBase::SendCommand(unsigned char, chip::System::PacketBufferHandle, chip::Callback::Cancelable*, chip::Callback::Cancelable*) + 152 (CHIPCluster.cpp:55)
 11  CHIP                          	 0x00000001ae464f14 chip::Controller::BindingCluster::Bind(chip::Callback::Cancelable*, chip::Callback::Cancelable*, unsigned long long, unsigned short, unsigned char, unsigned short) + 160 (CHIPClusters.cpp:402)
 12  CHIP                          	 0x00000001ae3e4bcc -[CHIPBinding bind:groupId:endpointId:clusterId:completionHandler:] + 840 (CHIPClustersObjc.mm:2343)
```

Logs
```
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔵 CHIP: [IN] Sending msg from 0x9db474590cb1a5acto 0x000000000000000d
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔵 CHIP: [IN] Sending secure msg on generic transport
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔵 CHIP: [IN] Secure msg send status 2009
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔵 CHIP: [IN] Sending msg from 0x9db474590cb1a5acto 0x000000000000000d
🔵 CHIP: [IN] Sending secure msg on generic transport
🔵 CHIP: [IN] Secure msg send status 2064
🔴 CHIP: [CTL] Failed in sending cluster command. Err 2064
🔵 CHIP: [IN] Sending msg from 0x9db474590cb1a5acto 0x0000000000000005
🔵 CHIP: [IN] Sending secure msg on generic transport
🔵 CHIP: [IN] Secure msg send status 2064
🔴 CHIP: [CTL] Failed in sending cluster command. Err 2064
🔵 CHIP: [IN] Failed to initialize Udp transport: Inet Error 1003 (0x000003EB): Incorrect state
🔴 CHIP: [CTL] LoadSecureSessionParameters returning error 1003
🔴 CHIP: [CTL] Failed in sending cluster command. Err 1003
```

woody-apple, 2021-04-14:
@vivien-apple / @bzbarsky-apple could you take a peek?

bzbarsky-apple, 2021-04-28:
My best guess is that this is fundamentally the same as https://github.com/project-chip/connectedhomeip/issues/5963

woody-apple, 2022-02-07:
Closing, will re-open if still is present.

closed 2022-02-07 by woody-apple