project-chip/connectedhomeip #5963: Crash in chip::System::Object::Release:280

woody-apple, 2021-04-12:
#### Problem
Crash here:
```
 Thread 15 Crashed:
 0   libsystem_kernel.dylib        	 0x00000001897f31e0 __pthread_kill + 8
 1   libsystem_pthread.dylib       	 0x000000018988b838 pthread_kill + 268 (pthread.c:1582)
 2   libsystem_c.dylib             	 0x0000000189758c24 abort + 164 (abort.c:118)
 3   CHIP                          	 0x00000001ae4bcf6c chip::System::Object::Release() + 88 (SystemObject.cpp:59)
 4   CHIP                          	 0x00000001ae48361c chip::Inet::UDPEndPoint::Free() + 40 (UDPEndPoint.cpp:446)
 5   CHIP                          	 0x00000001ae517d5c chip::Transport::UDP::Init(chip::Transport::UdpListenParameters&amp;) + 408 (UDP.cpp:70)
 6   CHIP                          	 0x00000001ae474f28 int chip::Transport::Tuple<chip::Transport::UDP, chip::Transport::UDP>::InitImpl<chip::Transport::UdpListenParameters&amp;>(chip::Transport::RawTransportDelegate*, chip::Transport::UdpListenParameters&amp;&amp;&amp;) + 76 (Tuple.h:203)
 7   CHIP                          	 0x00000001ae474ea8 int chip::Transport::Tuple<chip::Transport::UDP, chip::Transport::UDP>::InitImpl<chip::Transport::UdpListenParameters&amp;, chip::Transport::UdpListenParameters&amp;>(chip::Transport::RawTransportDelegate*, chip::Transport::UdpListenParameters&amp;&amp;&amp;, chip::Transport::UdpListenParameters&amp;&amp;&amp;) + 148 (Tuple.h:211)
 8   CHIP                          	 0x00000001ae474df4 int chip::Transport::Tuple<chip::Transport::UDP, chip::Transport::UDP>::Init<chip::Transport::UdpListenParameters&amp;, chip::Transport::UdpListenParameters&amp;, (void*)0>(chip::Transport::RawTransportDelegate*, chip::Transport::UdpListenParameters&amp;&amp;&amp;, chip::Transport::UdpListenParameters&amp;&amp;&amp;) + 88 (Tuple.h:106)
 9   CHIP                          	 0x00000001ae46f28c int chip::TransportMgr<chip::Transport::UDP, chip::Transport::UDP>::ResetTransport<chip::Transport::UdpListenParameters&amp;, chip::Transport::UdpListenParameters&amp;>(chip::Transport::UdpListenParameters&amp;&amp;&amp;, chip::Transport::UdpListenParameters&amp;&amp;&amp;) + 84 (TransportMgr.h:77)
 10  CHIP                          	 0x00000001ae46e480 chip::Controller::Device::LoadSecureSessionParameters(chip::Controller::Device::ResetTransport) + 264 (CHIPDevice.cpp:325)
 11  CHIP                          	 0x00000001ae46e164 chip::Controller::Device::SendMessage(chip::System::PacketBufferHandle, chip::PayloadHeader&amp;) + 380 (CHIPDevice.cpp:89)
 12  CHIP                          	 0x00000001ae46e65c chip::Controller::Device::SendMessage(chip::System::PacketBufferHandle) + 84 (CHIPDevice.cpp:138)
 13  CHIP                          	 0x00000001ae4636b0 chip::Controller::ClusterBase::SendCommand(unsigned char, chip::System::PacketBufferHandle, chip::Callback::Cancelable*, chip::Callback::Cancelable*) + 152 (CHIPCluster.cpp:55)
 14  CHIP                          	 0x00000001ae464f14 chip::Controller::BindingCluster::Bind(chip::Callback::Cancelable*, chip::Callback::Cancelable*, unsigned long long, unsigned short, unsigned char, unsigned short) + 160 (CHIPClusters.cpp:402)
 15  CHIP                          	 0x00000001ae3e4bcc -[CHIPBinding bind:groupId:endpointId:clusterId:completionHandler:] + 840 (CHIPClustersObjc.mm:2343)
```

Logs
```
🔵 CHIP: [IN] Sending msg from 0x9db474590cb1a5acto 0x0000000000000011
🔵 CHIP: [IN] Sending secure msg on generic transport
🔵 CHIP: [IN] Secure msg send status 2064
🔵 CHIP: [IN] Sending msg from 0x9db474590cb1a5acto 0x0000000000000011
🔵 CHIP: [IN] Sending secure msg on generic transport
🔵 CHIP: [IN] Secure msg send status 2064
🔴 CHIP: [CTL] Failed in sending cluster command. Err 2064
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔵 CHIP: [IN] Failed to initialize Udp transport: Inet Error 1003 (0x000003EB): Incorrect state
🔵 CHIP: [IN] Failed to initialize Udp transport: OS Error 2009 (0x000007D9): Bad file descriptor
```

woody-apple, 2021-04-14:
@vivien-apple / @bzbarsky-apple could you take a peek?

bzbarsky-apple, 2021-04-28:
Well, either we have memory corruption or we have some sort of weird reentry (perhaps from multiple threads?) into `UDP::Init`.  The crash is happening because `mUDPEndPoint` is not null so we are trying to free it, but it also has a zero refcount, so our attempt to release it hits the abort() in `Object::Release`.

The two consecutive "Failed to initialize Udp transport" logs strongly points towards the reentrancy situation as far as I am concerned.  @vivien-apple or @sagar-apple might have even fixed something like this recently, iirc....

closed 2022-01-25 by woody-apple