project-chip/connectedhomeip #5964: Crash in chip::Inet::IPEndPointBasis::GetSocket:204

woody-apple, 2021-04-12:
#### Problem
Crash here:
```
 Thread 11 Crashed:
 0   CHIP                          	 0x00000001ae47c038 chip::Inet::IPEndPointBasis::GetSocket(chip::Inet::IPAddressType, int, int) + 44 (IPEndPointBasis.cpp:954)
 1   CHIP                          	 0x00000001ae48349c chip::Inet::UDPEndPoint::GetSocket(chip::Inet::IPAddressType) + 56 (UDPEndPoint.cpp:923)
 2   CHIP                          	 0x00000001ae48349c chip::Inet::UDPEndPoint::GetSocket(chip::Inet::IPAddressType) + 56 (UDPEndPoint.cpp:923)
 3   CHIP                          	 0x00000001ae483788 chip::Inet::UDPEndPoint::SendMsg(chip::Inet::IPPacketInfo const*, chip::System::PacketBufferHandle, unsigned short) + 76 (UDPEndPoint.cpp:656)
 4   CHIP                          	 0x00000001ae518008 chip::Transport::UDP::SendMessage(chip::PacketHeader const&amp;, chip::Transport::PeerAddress const&amp;, chip::System::PacketBufferHandle) + 272 (UDP.cpp:105)
 5   CHIP                          	 0x00000001ae4749cc int chip::Transport::Tuple<chip::Transport::UDP, chip::Transport::UDP>::SendMessageImpl<1ul, (void*)0>(chip::PacketHeader const&amp;, chip::Transport::PeerAddress const&amp;, chip::System::PacketBufferHandle&amp;&amp;) + 240 (Tuple.h:172)
 6   CHIP                          	 0x00000001ae4748a0 int chip::Transport::Tuple<chip::Transport::UDP, chip::Transport::UDP>::SendMessageImpl<0ul, (void*)0>(chip::PacketHeader const&amp;, chip::Transport::PeerAddress const&amp;, chip::System::PacketBufferHandle&amp;&amp;) + 304 (Tuple.h:174)
 7   CHIP                          	 0x00000001ae474380 chip::Transport::Tuple<chip::Transport::UDP, chip::Transport::UDP>::SendMessage(chip::PacketHeader const&amp;, chip::Transport::PeerAddress const&amp;, chip::System::PacketBufferHandle) + 84 (Tuple.h:86)
 8   CHIP                          	 0x00000001ae4ca358 chip::TransportMgrBase::SendMessage(chip::PacketHeader const&amp;, chip::Transport::PeerAddress const&amp;, chip::System::PacketBufferHandle&amp;&amp;) + 156 (TransportMgrBase.cpp:28)
 9   CHIP                          	 0x00000001ae4c806c chip::SecureSessionMgr::SendMessage(chip::SecureSessionHandle, chip::PayloadHeader&amp;, chip::PacketHeader&amp;, chip::System::PacketBufferHandle, chip::EncryptedPacketBufferHandle*, chip::SecureSessionMgr::EncryptionState) + 984 (SecureSessionMgr.cpp:215)
 10  CHIP                          	 0x00000001ae4c7c44 chip::SecureSessionMgr::SendMessage(chip::SecureSessionHandle, chip::PayloadHeader&amp;, chip::System::PacketBufferHandle&amp;&amp;, chip::EncryptedPacketBufferHandle*) + 136 (SecureSessionMgr.cpp:122)
 11  CHIP                          	 0x00000001ae4c7bb0 chip::SecureSessionMgr::SendMessage(chip::SecureSessionHandle, chip::System::PacketBufferHandle&amp;&amp;) + 96 (SecureSessionMgr.cpp:115)
 12  CHIP                          	 0x00000001ae46e1b4 chip::Controller::Device::SendMessage(chip::System::PacketBufferHandle, chip::PayloadHeader&amp;) + 460 (CHIPDevice.cpp:91)
 13  CHIP                          	 0x00000001ae46e65c chip::Controller::Device::SendMessage(chip::System::PacketBufferHandle) + 84 (CHIPDevice.cpp:138)
 14  CHIP                          	 0x00000001ae4636b0 chip::Controller::ClusterBase::SendCommand(unsigned char, chip::System::PacketBufferHandle, chip::Callback::Cancelable*, chip::Callback::Cancelable*) + 152 (CHIPCluster.cpp:55)
 15  CHIP                          	 0x00000001ae464f14 chip::Controller::BindingCluster::Bind(chip::Callback::Cancelable*, chip::Callback::Cancelable*, unsigned long long, unsigned short, unsigned char, unsigned short) + 160 (CHIPClusters.cpp:402)
 16  CHIP                          	 0x00000001ae3e4bcc -[CHIPBinding bind:groupId:endpointId:clusterId:completionHandler:] + 840 (CHIPClustersObjc.mm:2343)
```

Logs
```
🔵 CHIP: [IN] Sending secure msg on generic transport
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔵 CHIP: [IN] Secure msg send status 2009
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔴 CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor
🔵 CHIP: [IN] Sending msg from 0x9db474590cb1a5acto 0x000000000000000c
🔵 CHIP: [IN] Sending secure msg on generic transport
```

woody-apple, 2021-04-14:
@vivien-apple / @bzbarsky-apple could you take a peek?

bzbarsky-apple, 2021-04-28:
This is crashing on what I assume is the attempt to read the `mSocket` member of `IPEndPointBasis`.  

If we had a crash address we could perhaps say more, like whether `UDP::mUDPEndpoint` is null or just garbage....

closed 2022-01-25 by woody-apple