project-chip/connectedhomeip #5965: Crash in int chip::app::CHIPDeviceCallbacksMgr::GetCallback

woody-apple, 2021-04-12:
#### Problem
Crash:
```
 Thread 11 Crashed:
 0   CHIP                          	 0x00000001e1a33300 int chip::app::CHIPDeviceCallbacksMgr::GetCallback<(anonymous namespace)::ReportCallbackInfo>((anonymous namespace)::ReportCallbackInfo&amp;, chip::Callback::CallbackDeque&amp;, chip::Callback::Cancelable**) + 60 (CHIPDeviceCallbacksMgr.h:80)
 1   CHIP                          	 0x00000001e1a3320c int chip::app::CHIPDeviceCallbacksMgr::CancelCallback<(anonymous namespace)::ReportCallbackInfo>((anonymous namespace)::ReportCallbackInfo&amp;, chip::Callback::CallbackDeque&amp;) + 52 (CHIPDeviceCallbacksMgr.h:67)
 2   CHIP                          	 0x00000001e1a3320c int chip::app::CHIPDeviceCallbacksMgr::CancelCallback<(anonymous namespace)::ReportCallbackInfo>((anonymous namespace)::ReportCallbackInfo&amp;, chip::Callback::CallbackDeque&amp;) + 52 (CHIPDeviceCallbacksMgr.h:67)
 3   CHIP                          	 0x00000001e1a331b4 chip::app::CHIPDeviceCallbacksMgr::AddReportCallback(unsigned long long, unsigned char, unsigned short, unsigned short, chip::Callback::Cancelable*) + 132 (CHIPDeviceCallbacksMgr.cpp:104)
 4   CHIP                          	 0x00000001e1a3f3e0 chip::Controller::Device::AddReportHandler(unsigned char, unsigned short, unsigned short, chip::Callback::Cancelable*) + 68 (CHIPDevice.cpp:366)
 5   CHIP                          	 0x00000001e1a337ec chip::Controller::ClusterBase::RequestAttributeReporting(unsigned short, chip::Callback::Cancelable*) + 92 (CHIPCluster.cpp:77)
 6   CHIP                          	 0x00000001e1a3d110 chip::Controller::OnOffCluster::ReportAttributeOnOff(chip::Callback::Cancelable*) + 40 (CHIPClusters.cpp:3963)
```

Logs
```
🔵 CHIP: [ZCL] RX len 4, ep 1, clus 0x 6
🔵 CHIP: [ZCL]  FC 8 seq 2 cmd 7 payload[
🔵 CHIP: [ZCL] 00
🔵 CHIP: [ZCL] ]
🔵 CHIP: [ZCL] ConfigureReportingResponseCallback:
🔵 CHIP: [ZCL]   ClusterId: 0x0006
🔵 CHIP: [ZCL]   status: EMBER_ZCL_STATUS_SUCCESS (0x00)
```

woody-apple, 2021-04-14:
@vivien-apple / @bzbarsky-apple could you take a peek?

vivien-apple, 2021-04-15:
Assigning to myself for now.

vivien-apple, 2021-04-15:
What I do not understand in the stack at the moment are why do we have:
```
 1   CHIP                          	 0x00000001e1a3320c int chip::app::CHIPDeviceCallbacksMgr::CancelCallback<(anonymous namespace)::ReportCallbackInfo>((anonymous namespace)::ReportCallbackInfo&amp;, chip::Callback::CallbackDeque&amp;) + 52 (CHIPDeviceCallbacksMgr.h:67)
 2   CHIP                          	 0x00000001e1a3320c int chip::app::CHIPDeviceCallbacksMgr::CancelCallback<(anonymous namespace)::ReportCallbackInfo>((anonymous namespace)::ReportCallbackInfo&amp;, chip::Callback::CallbackDeque&amp;) + 52 (CHIPDeviceCallbacksMgr.h:67)
```

vivien-apple, 2021-04-15:
The stack suggest that someone is trying to add a report handler in the `ConfigureReportingResponseCallback`, which sounds perfectly legitimate.

closed 2022-01-25 by woody-apple