project-chip/connectedhomeip #6023: SRP Client on the accessory used incorrect srp server IP address

cjandhyala, 2021-04-14:
This is an issue with OTBR SRP client , tracked with OTBR issue https://github.com/openthread/openthread/issues/6447

I: 50507 [DL]SRP Client was started, as detected server addressed: fdf2:7a68:5f3e:108c:0:ff:fe00:fc11
D: 50517 [DL]OpenThread State Changed (Flags: 0x301332b7)
D: 50522 [DL]   Device Role: CHILD
D: 50525 [DL]   Network Name: MyHomeBR
D: 50528 [DL]   PAN Id: 0xABCD
D: 50531 [DL]   Extended PAN Id: 0x949CD1F473FF7008
D: 50536 [DL]   Channel: 11
D: 50538 [DL]   Mesh Prefix: fdf2:7a68:5f3e:108c::/64
D: 50543 [DL]   Partition Id: 0x33AA34DE
D: 50547 [DL]   Thread Unicast Addresses:
D: 50551 [DL]        fd31:296e:280b:0:fd04:43b4:5bc1:85ca/64 valid preferred
D: 50558 [DL]        fdf2:7a68:5f3e:108c:0:ff:fe00:c04/64 valid preferred rloc
D: 50565 [DL]        fdf2:7a68:5f3e:108c:c41c:d4a5:3f62:b55e/64 valid preferred
D: 50572 [DL]        fe80::14fd:46fd:1be9:34db/64 valid preferred
I: 50578 [NP]Sending IP Address. Delegate 0x20005b8c

D: 50583 [IN]Secure message was encrypted: Msg ID 0
I: 50588 [IN]Sending msg from 0x0000000000bc5c01 to 0x0000000000000000 at utc time: 50588 msec
I: 50596 [IN]Sending secure msg on connection specific transport
D: 50602 [DL]Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
I: 50609 [IN]Secure msg send status No Error
D: 50613 [DL]OpenThread State Changed (Flags: 0x00000001)
D: 50619 [DL]   Thread Unicast Addresses:
D: 50622 [DL]        fd31:296e:280b:0:fd04:43b4:5bc1:85ca/64 valid preferred
D: 50629 [DL]        fdf2:7a68:5f3e:108c:0:ff:fe00:c04/64 valid preferred rloc
D: 50636 [DL]        fdf2:7a68:5f3e:108c:c41c:d4a5:3f62:b55e/64 valid preferred
D: 50643 [DL]        fe80::14fd:46fd:1be9:34db/64 valid preferred
I: 50649 [NP]Sending IP Address. Delegate 0x20005b8c

D: 50655 [IN]Secure message was encrypted: Msg ID 1
I: 50659 [IN]Sending msg from 0x0000000000bc5c01 to 0x0000000000000000 at utc time: 50659 msec
I: 50668 [IN]Sending secure msg on connection specific transport
I: 50673 [IN]Secure msg send status No Error
D: 50796 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 50804 [DL]Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 50811 [DL]Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 50891 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 50899 [DL]Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 50988 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 50996 [DL]Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 51003 [DL]Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 51085 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 51093 [DL]Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 9)
D: 51183 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 51191 [DL]Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 51198 [DL]Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 51281 [DL]ConnId: 0x00, New CCCD value: 0x0000
D: 51285 [DL]SetAdvertisingEnabled(false)
I: 51289 [SVR]Persisting admin ID 0, next available 1
I: 51481 [SVR]Persisting admin ID successfully
I: 51486 [SVR]OnRendezvousError: Ble Error 6004 (0x00001774): BLE central unsubscribed
I: 51494 [BLE]Releasing end point's BLE connection back to application.
E: 51500 [DL]Long dispatch time: 219 ms
D: 51504 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
E: 51511 [BLE]no endpoint for BLE sent data ack
D: 51516 [DL]In DriveBLEState
E: 52593 [DL]OnSrpClientNotification: Timed out waiting on server response
I: 53669 [DL]BLE GAP connection terminated (reason 0x13)
I: 53674 [DL]Current number of connections: 0/1
D: 53678 [DL]In DriveBLEState
E: 55686 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 60921 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 69794 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 84856 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 110423 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 153858 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 227671 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 353131 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 566380 [DL]OnSrpClientNotification: Timed out waiting on server response

franck-apple, 2021-05-17:
openthread/openthread#6447 is resolved. Can we close this issue?

Damian-Nordic, 2021-05-18:
@franck-apple we still need to bump OpenThread versions in all platforms.

Damian-Nordic, 2021-07-21:
It's resolved already.

closed 2021-07-21 by Damian-Nordic