project-chip/connectedhomeip #6090: dhclient command returns error inside docker container

cjandhyala, 2021-04-16:
dhclient command run inside docker container (accessory) during provisioning returns error.  this is required for the accessory to get the IP address

CHIP:DL: wpa_supplicant: save config succeeded!
CHIP:DL: Failed to run dhclient, system() returns 32512
CHIP:SVR: Device was assigned network credentials
CHIP:DL: MDNS failed to join multicast group on wlan0 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
CHIP:DIS: CHIP minimal mDNS started advertising.
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Found admin paring for admin 0, node BC5C01
CHIP:DIS: Using wifi MAC for hostname
CHIP:DIS: CHIP minimal mDNS configured as 'Operational device'.
CHIP:SVR: Device completed Rendezvous process
CHIP:SVR: Failed to store the connection state
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 517
CHIP:BLE: Write request received debug 0xffff7404c1b0
CHIP:BLE: Indication confirmation, 0xffff7404c1b0

erjiaqing, 2021-04-19:
Can you try to add `--cap-add=net_admin` when bringing up container?

Seems this is not an issue of our code, but some limitations about docker.

cjandhyala, 2021-04-19:
@erjiaqing , I have it in the command Song, 
docker run --name dockercontroller -v /var/run/dbus:/var/run/dbus --privileged --cap-add=SYS_ADMIN --cap-add=NET_ADMIN --net=host -it chip-stack-image