project-chip/connectedhomeip #6094: erro Unresolved reference: chip

hainahouwei, 2021-04-16:
#### Problem
<what's wrong or missing, please include any applicable:
 * Click Make Project in Android Studio to build Android CHIPTool
 * Unresolved reference: chip
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

Damian-Nordic, 2021-04-16:
@hainahouwei Did you follow the steps described in https://github.com/project-chip/connectedhomeip/tree/master/src/android/CHIPTool? It seems you haven't built the CHIP controller JAR before builting the CHIPTool project.

AdamZhang0124, 2021-07-02:
Hi @Damian-Nordic @hainahouwei ,
I have encountered the same problem, therefore I would like to check if this issue has any progress?
I do have followed the the steps described in https://github.com/project-chip/connectedhomeip/tree/master/src/android/CHIPTool, but still, the build is failed.

Attach the screenshot fyi:
[image]

I noticed that you mentioned **the CHIP controller JAR** , may I know which file/command you were referring to, I did not see it on the README.md?
Many thanks!

austinh0, 2021-07-08:
`scripts/examples/android_app.sh` should generate `CHIPController.jar` and `SetupPayloadParser.jar`, and copy them to `src/android/CHIPTool/app/libs`.

It looks like you're missing these jars, resulting in the unresolved reference errors. Are you running `scripts/examples/android_app.sh` before you sync/make? Can you check if those jars are there?

jonsmirl, 2021-08-25:
I just ran into this also. The problem is in the Android build instructions. If the environment is not bootstrapped the android_app.sh script fails.

1) Check out the CHIP repository.

>>>> 1a) Bootstrap it.  source scripts/bootstrap.sh

2) In the command line, run the following command from the top CHIP directory:

TARGET_CPU=arm64 ./scripts/examples/android_app.sh
See the table above for other values of TARGET_CPU.

3) Open the project in Android Studio and run Sync Project with Gradle Files.

Use one of the following options to build an Android package:

4)  Click Make Project in Android Studio.

Run the following command in the command line:

cd src/android/CHIPTool
./gradlew build

austinh0, 2021-08-25:
Makes sense, bootstrap is required before we build anything in CHIP. I will update the docs.

asokfair, 2022-10-01:
-->>>> 1a) Bootstrap it. source scripts/bootstrap.sh gives us error, 
[image]

VinhBH1, 2024-06-04:
Hi @austinh0 
where is the location of CHIPController.jar and SetupPayloadParser.jar

 i have follow in https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md
but still not see  CHIPController.jar and SetupPayloadParser.jar

BlackLordj, 2024-12-30:
> CHÀO@austinh0 vị trí của CHIPController.jar và SetupPayloadParser.jar ở đâu
> 
> tôi đã theo dõi trong https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md nhưng vẫn không thấy CHIPController.jar và SetupPayloadParser.jar
I'm also having the same error as you, I've run the command TARGET_CPU=x64 ./scripts/examples/android_app_ide.sh many times but it still doesn't have .jar files in the android/CHIPtool/app/lib/ folder. Do you have a solution yet?

BlackLordj, 2024-12-30:
> ```
> jonsmirl@ares:~/aosp/esp-matter/connectedhomeip/connectedhomeip$ ls ./examples/android/CHIPTool/app/libs
> AndroidPlatform.jar  CHIPClusters.jar    jniLibs            libMatterTlv.jar       README.md
> CHIPClusterID.jar    CHIPController.jar  libMatterJson.jar  OnboardingPayload.jar
> ```
I still have the problem that it doesn't create .jar files after running the set_target command