project-chip/connectedhomeip #6110: chip shell coredumps on sending ping command

cjandhyala, 2021-04-16:
chip shell coredumps on sending ping command
$ ./out/debug/linux_arm64_gcc/chip-shell
Init CHIP Stack
Init Thread stack
CHIP:DL: ../../src/platform/Linux/ThreadStackManagerImpl.cpp 94: Otbr ClientError 23
ThreadStackMgr().InitThreadStack() failed
> CHIP:DL: CHIP task running
> ping -r 1 192.168.4.244
IP address: 192.168.4.244
CHIP:IN: TransportMgr initialized
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x000000000001b669
CHIP:IN: New pairing for device 0x0000000000bc5c01, key 0!!
Establish secure session succeeded
Send echo request message with payload size: 32 bytes to Node: 12344321
CHIP:IN: Secure message was encrypted: Msg ID 0
CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000bc5c01 at utc time: 57959404 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:EM: Rxd Ack; Removing MsgId:00000000 from Retrans Table
CHIP:EM: Removed CHIP MsgId:00000000 from RetransTable
CHIP:EM: Sending Standalone Ack for MsgId:00000001
CHIP:EM: Failed to send Solitary ack for MsgId:00000001:4003
Echo Response: 1/1(100.00%) len=32 time=0.002ms
CHIP:EM: Sending Standalone Ack for MsgId:00000001
Aborted (core dumped)

Log on the echo responder
$ ./out/debug/linux_arm64_gcc/chip-echo-responder
Init CHIP Stack
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x0000000000bc5c01
CHIP:IN: New pairing for device 0x000000000001b669, key 0!!
Listening for Echo requests...
CHIP:EM: ec pos: 0, id: 17767, Delegate: 0xb7ce9008
Echo Request, len=32 ... sending response.
CHIP:EM: Piggybacking Ack for MsgId:00000000 with msg
CHIP:IN: Secure message was encrypted: Msg ID 0
CHIP:IN: Sending msg from 0x0000000000bc5c01 to 0x000000000001b669 at utc time: 257334589 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:IN: Sending msg from 0x0000000000bc5c01 to 0x000000000001b669 at utc time: 257339568 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:EM: Retransmit MsgId:00000000 Send Cnt 1
CHIP:IN: Sending msg from 0x0000000000bc5c01 to 0x000000000001b669 at utc time: 257339820 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:EM: Retransmit MsgId:00000000 Send Cnt 2
CHIP:IN: Sending msg from 0x0000000000bc5c01 to 0x000000000001b669 at utc time: 257340077 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:EM: Retransmit MsgId:00000000 Send Cnt 3
CHIP:EM: Failed to Send CHIP MsgId:00000000 sendCount: 3 max retries: 3
CHIP:EM: ec pos: 0, id: 17767, Delegate: 0xb7ce9008
Echo Request, len=32 ... sending response.
CHIP:EM: Piggybacking Ack for MsgId:00000000 with msg
CHIP:IN: Secure message was encrypted: Msg ID 1
CHIP:IN: Sending msg from 0x0000000000bc5c01 to 0x000000000001b669 at utc time: 257480672 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:IN: Sending msg from 0x0000000000bc5c01 to 0x000000000001b669 at utc time: 257485618 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:EM: Retransmit MsgId:00000001 Send Cnt 1
CHIP:IN: Sending msg from 0x0000000000bc5c01 to 0x000000000001b669 at utc time: 257485868 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:EM: Retransmit MsgId:00000001 Send Cnt 2
CHIP:IN: Sending msg from 0x0000000000bc5c01 to 0x000000000001b669 at utc time: 257486125 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:EM: Retransmit MsgId:00000001 Send Cnt 3
CHIP:EM: Failed to Send CHIP MsgId:00000001 sendCount: 3 max retries: 3

yufengwangca, 2021-04-21:
Not able to reproduce this issue on the latest commit:  2265b34f59030e4039ab4d0aa4ac56901fc8c6f4

yufengw@yufengw-SEi:~/Workspace/connectedhomeip/out/host$ ./chip-shell 
Init CHIP Stack
Init Thread stack
CHIP:DL: ../../src/platform/Linux/ThreadStackManagerImpl.cpp 94: Otbr ClientError 23
ThreadStackMgr().InitThreadStack() failed
> CHIP:DL: CHIP task running

> 
> 
> ping -r 1 192.168.86.171
IP address: 192.168.86.171
CHIP:IN: TransportMgr initialized
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x000000000001b669
CHIP:IN: New pairing for device 0x0000000000bc5c01, key 0!!
Establish secure session succeeded

Send echo request message with payload size: 32 bytes to Node: 12344321
CHIP:IN: Secure message was encrypted: Msg ID 0
CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000bc5c01 at utc time: 1075750127 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:EM: Rxd Ack; Removing MsgId:00000000 from Retrans Table
CHIP:EM: Removed CHIP MsgId:00000000 from RetransTable
CHIP:EM: Sending Standalone Ack for MsgId:00000000
CHIP:IN: Secure message was encrypted: Msg ID 1
CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000bc5c01 at utc time: 1075750128 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:EM: Flushed pending ack for MsgId:00000000
Echo Response: 1/1(100.00%) len=32 time=0.002ms

Send echo request message with payload size: 32 bytes to Node: 12344321
CHIP:IN: Secure message was encrypted: Msg ID 2
CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000bc5c01 at utc time: 1075751127 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:EM: Rxd Ack; Removing MsgId:00000002 from Retrans Table
CHIP:EM: Removed CHIP MsgId:00000002 from RetransTable
CHIP:EM: Sending Standalone Ack for MsgId:00000001
CHIP:IN: Secure message was encrypted: Msg ID 3
CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000bc5c01 at utc time: 1075751128 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:EM: Flushed pending ack for MsgId:00000001
Echo Response: 2/2(100.00%) len=32 time=0.002ms

Send echo request message with payload size: 32 bytes to Node: 12344321
CHIP:IN: Secure message was encrypted: Msg ID 4
CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000bc5c01 at utc time: 1075752128 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:EM: Rxd Ack; Removing MsgId:00000004 from Retrans Table
CHIP:EM: Removed CHIP MsgId:00000004 from RetransTable
CHIP:EM: Sending Standalone Ack for MsgId:00000002
CHIP:IN: Secure message was encrypted: Msg ID 5
CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000bc5c01 at utc time: 1075752129 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:EM: Flushed pending ack for MsgId:00000002
Echo Response: 3/3(100.00%) len=32 time=0.001ms
Done
>

yufengwangca, 2021-04-21:
@cjandhyala

Could you retest on the ToT and see if you still see the issue, if so, please lets me know the commit of your branch and on which platform you run the test.

franck-apple, 2021-05-17:
@cjandhyala can we close this one?