project-chip/connectedhomeip #6118: python controller on raspi times out to connect to BLE accessory running on raspi

cjandhyala, 2021-04-19:
Occasionally python controller on raspi times out to connect to BLE accessory running on raspi.  

Both controller and accessory on commit ID 
commit 88780da9141b6bb4476f476fdd736590328378e0 (HEAD -> master, origin/master, origin/HEAD)
Author: Kevin Schoedel <67607049+kpschoedel@users.noreply.github.com>
Date:   Sat Apr 17 23:23:14 2021 -0400

    Remove `exit: return` idiom from src/inet and src/transport (#6025)
    
    * Remove `exit: return` idiom from src/inet and src/transport
    
    #### Problem

Controller side logs
-------------------
$ chip-device-ctrl 
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x000000000001b669
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:DL: CHIP task running
CHIP:DL: Platform main loop started.
Chip Device Controller Shell

chip-device-ctrl > set-pairing-wifi-credential Partyguys Invalid$123
WiFi credential set
chip-device-ctrl > connect -ble 3840 12345678 12344321
Device is assigned with nodeid = 12344321
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:DL: TRACE: Bus acquired for name C-360b
CHIP:DL: PlatformBlueZInit init success
CHIP:BLE: BLE removing known devices.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: Device 08:66:98:91:F5:27 does not look like a CHIP device.
CHIP:BLE: Device 1E:A2:77:BA:3B:F4 does not look like a CHIP device.
CHIP:BLE: Device 63:5D:DC:52:FF:A6 does not look like a CHIP device.
CHIP:BLE: Device 5A:8F:D0:02:02:44 does not look like a CHIP device.
CHIP:BLE: Device 61:BB:51:B3:0E:D4 does not look like a CHIP device.
CHIP:BLE: Device 60:D1:43:A6:2E:95 does not look like a CHIP device.
CHIP:BLE: Device 4F:3D:E4:8A:FB:F2 does not look like a CHIP device.
CHIP:BLE: Device 45:4A:CE:06:B7:7A does not look like a CHIP device.
CHIP:BLE: Device 3C:5E:B6:52:10:6B does not look like a CHIP device.
CHIP:BLE: Device 6D:CD:DD:64:0E:76 does not look like a CHIP device.
CHIP:BLE: Device 70:97:06:48:53:D7 does not look like a CHIP device.
CHIP:BLE: Device 77:36:05:2C:C0:48 does not look like a CHIP device.
CHIP:BLE: Device 00:6C:95:47:0F:B1 does not look like a CHIP device.
CHIP:BLE: Device EB:F8:6B:27:DB:FF does not look like a CHIP device.
CHIP:BLE: Device D1:34:C5:E6:77:D8 does not look like a CHIP device.
CHIP:BLE: Device 34:74:03:FC:0A:DB does not look like a CHIP device.
CHIP:BLE: Device 45:75:1F:BA:2E:F8 does not look like a CHIP device.
CHIP:BLE: New device scanned: DC:A6:32:8D:70:74
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Scan complete notification without an active scan.
CHIP:DL: HandlePlatformSpecificBLEEvent 16386
CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
Failed to establish secure session to device: 4050
CHIP:DL: FAIL: ConnectDevice : Operation was cancelled
CHIP:DL: HandlePlatformSpecificBLEEvent 16386
CHIP Error 4050 (0x00000FD2): Timeout
chip-device-ctrl > CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service00bb
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service00bb
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service00bb
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service00bb
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service00bb
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
[... 376 lines cut ...]
CHIP:DL: HandlePlatformSpecificBLEEvent 16395
CHIP:DL: CHIPoBLE advertising stopped
CHIP:DIS: Broadcasting mDns reply
CHIP:EM: ec pos: 1, id: 17769, Delegate: 0xb87afc10
CHIP:SVR: Packet received from Node:1b669: 14 bytes
CHIP:ZCL: APS frame processing success!
CHIP:ZCL: RX len 3, ep 1, clus 0x 6 
CHIP:ZCL:  FC 1 seq 0 cmd 2 payload[
CHIP:ZCL: NULL
CHIP:ZCL: ]
CHIP:ZCL: On/Off set value: 1 2
CHIP:ZCL: Toggle on/off from 0 to 1
LightingManager::InitiateAction(ON_ACTION)CHIP:ZCL: Unknown cluster ID: 8
CHIP:ZCL: Measured APS frame size 11
CHIP:ZCL: Successfully encoded 11 bytes
CHIP:IN: Secure message was encrypted: Msg ID 1
CHIP:IN: Sending msg from 0x0000000000bc5c01 to 0x000000000001b669 at utc time: 244605103 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:ZCL: T   0:TX (0xaaaab8780eb8) Ucast 0x00xaaaab8780eb0
CHIP:ZCL: TX buffer: [
CHIP:ZCL: 08 00 0B 02 00 
CHIP:ZCL: ]
CHIP:ZCL: Data model processing success!
CHIP:DIS: Broadcasting mDns reply

Damian-Nordic, 2021-04-22:
@cjandhyala Looks like Song Guo increased the ConnectDevice timeout from 5s to 10s in #5337. Let us know in a few days if you're still seeing this issue.

franck-apple, 2021-05-17:
@cjandhyala can we close this one?