project-chip/connectedhomeip #6121: Python controller to support Network Commissioning Cluster Commands ,

cjandhyala, 2021-04-19:
Python controller to support Network Commissioning Cluster Commands , Required for cert test case 
https://github.com/project-chip/chip-test-plans/blob/master/src/devicemanagement.adoc#6327-test-procedure 

[image]
[image]

Damian-Nordic, 2021-04-20:
@erjiaqing I can add support for the Network Commissioning Cluster commands to Python CHIP controller if you haven't started looking into that.

erjiaqing, 2021-04-20:
> @erjiaqing I can add support for the Network Commissioning Cluster commands to Python CHIP controller if you haven't started looking into that.

Seems no special support is required?
The network commissioning cluster is already in the zap file for controller.

Damian-Nordic, 2021-04-20:
I think we need to add Network Commissioning Cluster to the Python app's ZAP configuration and re-gen files in `python/gen` to support parsing responses to some of the commands.

PR #6205 (2021-04-22): [python] Process responses to Network Commissioning Cluster commands

 #### Problem
ZCL code for Python CHIP Controller doesn't process responses to Network Commissioning Cluster commands as the cluster is disabled in the ZAP configuration.

 #### Summary of Changes
Enable Network Commissioning Cluster and all commands in the ZAP configuration of Python CHIP Controller

**Note: only local-clusters.zap file needs to be reviewd. The rest it auto-generated!**

 Fixes #6121
closed 2021-04-22 by andy31415