project-chip/connectedhomeip #6123: Segmentation fault with EnableNetwork command in NetworkCommissioning using python controller

cjandhyala, 2021-04-19:
Seg fault with EnableNetwork command in NetworkCommissioning using python controller
 $ chip-device-ctrl 
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x000000000001b669
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:DL: CHIP task running
CHIP:DL: Platform main loop started.
Chip Device Controller Shell

chip-device-ctrl > help

Available commands (type help <name> for more information):
============================================================
ble-adapter-print   ble-scan  set-pairing-thread-credential  zcl    
ble-adapter-select  connect   set-pairing-wifi-credential    zclread
ble-debug-log       resolve   setup-payload                

chip-device-ctrl > set-pairing-wifi-credential TESTSSID PASSWD
WiFi credential set
chip-device-ctrl > connect -ble 3840 12345678 12344321
Device is assigned with nodeid = 12344321
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:DL: TRACE: Bus acquired for name C-a7d3
CHIP:DL: PlatformBlueZInit init success
CHIP:BLE: BLE removing known devices.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: Device 7B:2D:75:85:0F:73 does not look like a CHIP device.
CHIP:BLE: Device 08:66:98:91:F5:27 does not look like a CHIP device.
CHIP:BLE: Device 0E:BC:A0:27:F8:88 does not look like a CHIP device.
CHIP:BLE: Device 4C:B3:EF:82:82:4E does not look like a CHIP device.
CHIP:BLE: Device 79:DD:1C:CB:C8:88 does not look like a CHIP device.
CHIP:BLE: Device 32:1F:F5:0D:A4:E9 does not look like a CHIP device.
CHIP:BLE: Device 23:D4:77:99:9F:E4 does not look like a CHIP device.
CHIP:BLE: Device 77:36:05:2C:C0:48 does not look like a CHIP device.
CHIP:BLE: Device 1E:A2:77:BA:3B:F4 does not look like a CHIP device.
CHIP:BLE: Device FB:67:AC:AE:05:8A does not look like a CHIP device.
CHIP:BLE: Device 6A:55:F0:28:2C:BC does not look like a CHIP device.
CHIP:BLE: Device 6A:96:27:E2:D2:DC does not look like a CHIP device.
CHIP:BLE: Device 71:82:E7:CE:74:BB does not look like a CHIP device.
CHIP:BLE: Device 68:C7:C2:BD:71:DB does not look like a CHIP device.
CHIP:BLE: Device 5D:F5:12:02:E1:87 does not look like a CHIP device.
CHIP:BLE: Device 76:EC:48:07:B1:6C does not look like a CHIP device.
CHIP:BLE: Device 5C:B0:77:1B:02:CD does not look like a CHIP device.
CHIP:BLE: New device scanned: DC:A6:32:8D:70:74
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Scan complete notification without an active scan.
CHIP:DL: ConnectDevice complete
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service011b
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service011b
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service011b
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service011b
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service011b
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service011b
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service011b
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service011b
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service011b
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service011b
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service011b
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0010
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service011b
[... 359 lines cut ...]
CHIP:DL: wpa_supplicant: removed network: /fi/w1/wpa_supplicant1/Interfaces/0/Networks/16
CHIP:DL: wpa_supplicant: added network: SSID: Partyguys: /fi/w1/wpa_supplicant1/Interfaces/0/Networks/16
CHIP:DL: wpa_supplicant: connected to network: SSID: Partyguys
CHIP:DL: wpa_supplicant: save config succeeded!
CHIP:DL: Got IP address on interface: wlan0 IP: 192.168.6.94
CHIP:DL: dhclient is running on the wlan0 interface.
CHIP:ZCL: Measured APS frame size 11
CHIP:ZCL: Successfully encoded 11 bytes
CHIP:IN: Secure message was encrypted: Msg ID 2
CHIP:IN: Sending msg from 0x0000000000bc5c01 to 0x000000000001b669 at utc time: 269800047 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:ZCL: T   0:TX (0xaaaac6480eb8) Ucast 0x00xaaaac6480eb0
CHIP:ZCL: TX buffer: [
CHIP:ZCL: 08 01 0B 0C 00 
CHIP:ZCL: ]
CHIP:ZCL: Data model processing success!
CHIP:DL: HandlePlatformSpecificBLEEvent 32770
CHIP:NP: Sending IP Address

CHIP:IN: Secure message was encrypted: Msg ID 3
CHIP:IN: Sending msg from 0x0000000000bc5c01 to 0x000000000001b669 at utc time: 269800048 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:DIS: Broadcasting mDns reply

erjiaqing, 2021-05-12:
Now, we are using IM for sending messages, does this issue occurs on IM also?

cjandhyala, 2021-05-13:
@erjiaqing we don't see this issue anymore. we can close the issue and open a new issue when we see it again

closed 2021-05-13 by cjandhyala