project-chip/connectedhomeip #6145: CRMP ping aborts

cjandhyala, 2021-04-19:
CRMP ping aborts

> ping -c 1 -r 1 192.168.4.244
IP address: 192.168.4.244
CHIP:IN: TransportMgr initialized
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x000000000001b669
CHIP:IN: New pairing for device 0x0000000000bc5c01, key 0!!
Establish secure session succeeded

Send echo request message with payload size: 32 bytes to Node: 12344321
CHIP:IN: Secure message was encrypted: Msg ID 0
CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000bc5c01 at utc time: 306583575 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:EM: Rxd Ack; Removing MsgId:00000000 from Retrans Table
CHIP:EM: Removed CHIP MsgId:00000000 from RetransTable
CHIP:EM: Sending Standalone Ack for MsgId:00000000
CHIP:EM: Failed to send Solitary ack for MsgId:00000000:4003
Echo Response: 1/1(100.00%) len=32 time=0.002ms
CHIP:EM: Sending Standalone Ack for MsgId:00000000
Aborted (core dumped)

pan-apple, 2021-04-21:
@cjandhyala , could you provide the stack trace for the crash?

cjandhyala, 2021-04-21:
@pan-apple I don't see this issue anymore with ToT. will get stack trace if we see it again

closed 2021-04-21 by cjandhyala