project-chip/connectedhomeip #6159: Thread provisioning fails for Nordic dev board running lighting-app

emontnemery, 2021-04-20:
#### Problem
 * Expected behavior
   * The board running the lighting example should join the network and be possible to control
 * Actual behavior
   * Provisioning fails
 
 * Steps to reproduce
Follow this guide, https://github.com/project-chip/connectedhomeip/blob/master/examples/lighting-app/nrfconnect/README.md#device-ui, to build and flash the lighting-app to a second nRF52850 DK
I followed the "Using Docker container for setup" path.
Follow this guide to set up an nRF52840 as border router, build the Android CHIPTool and attempt to commision the board running the nRF Connect Lighting Example.
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/nrfconnect_android_commissioning.md

After choosing "PROVISION CHIP DEVICE WITH THREAD" in the CHIPTool app and scanning the device's QR code, LED1 starts blinking fast indicating that _The device is in the unprovisioned state and a commissioning application is connected through Bluetooth LE._
A few seconds later, the device returns to slow flash.

 * system configuration
   * On commit 2d43a7f26f8b217e8bc4440cc6e2526602d1a7cf
 * logs
 USB log from the board running the lighting-app:
```
I: 1992994 [DL]Current number of connections: 1/1
D: 1992998 [DL]In DriveBLEState
I: 1993001 [DL]CHIPoBLE advertising stopped
D: 1993817 [DL]Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
I: 1993825 [BLE]local and remote recv window sizes = 3
I: 1993830 [BLE]selected BTP version 3
I: 1993833 [BLE]using BTP fragment sizes rx 20 / tx 20.
D: 2008824 [DL]ConnId: 0x00, New CCCD value: 0x0000
E: 2008838 [BLE]receive handshake timed out, closing ep 0x200062e0
I: 2008844 [BLE]Releasing end point's BLE connection back to application.
I: 2009896 [DL]BLE GAP connection terminated (reason 0x13)
I: 2009901 [DL]Current number of connections: 0/1
D: 2009906 [DL]In DriveBLEState
I: 2009910 [DL]CHIPoBLE advertising started
I: 2039915 [DL]CHIPoBLE advertising mode changed to slow
D: 2039920 [DL]In DriveBLEState
D: 2909914 [DL]SetAdvertisingEnabled(false)
I: 2909918 [DL]CHIPoBLE advertising disabled because of timeout expired
D: 2909925 [DL]In DriveBLEState
I: 2909928 [DL]CHIPoBLE advertising stopped
```

 #### Proposed Solution
<suggested fix, suggested enhancement>

kkasperczyk-no, 2021-04-20:
Thank you for reporting that. We are aware of that problem and unfortunately currently communication between CHIP device and Android platform doesn't seem to work properly (at all, not only on nrf platform). That is because of the changes related to the rendezvous and provisioning on the master branch, that were not introduced in all platforms, but mainly in the Python CHIP device controller tool, which is used as the main certification and testing tool. That is definitely regression and problems with regular maintenance of all platforms, but hopefully as a community we plan to focus on bringing back to work all platforms (including mobile ones) again in the upcoming weeks. Unfortunately for now I can only recommend using Python CHIP device controller, as the most mature tool.

kkasperczyk-no, 2021-04-20:
In case you really want to test nRF DK with Android platform, I can recommend you trying to do this using following Nordic blogpost: https://devzone.nordicsemi.com/nordic/nordic-blog/b/blog/posts/connected-home-over-ip-sneak-peek-of-early-development-on-nrf52840 and commit revisions described there, however it was based on much older version that current master branch and will be missing plenty of new features.

emontnemery, 2021-04-21:
@kkasperczyk-no thanks a lot for the quick feedback, highly appreciated!