project-chip/connectedhomeip #6201: Add support for NetworkCommissioning cluster ScanNetworks command.

kkasperczyk-no, 2021-04-21:
#### Problem
Currently not all NetworkCommissioning cluster commands are supported on the accessory side (device responds with default not supported messages), including ScanNetworks command. This issue is also tracked in the more general issue considering full NetworkCommissioning cluster support.

 #### Proposed Solution
Add implementation for emberAfNetworkCommissioningClusterScanNetworksCallback() method.

arunbharadwaj, 2022-02-11:
Is ScanNetworks command now fully supported, or is this item still open?

kkasperczyk-no, 2022-02-14:
@arunbharadwaj I don't know, as I stopped working on that item long time ago, so I'm not aware of the current state. I believe that maybe @erjiaqing could be able to answer that.

closed 2022-06-27 by bzbarsky-apple