project-chip/connectedhomeip #6240: node id is all 0s in mDns advertisements

cjandhyala, 2021-04-22:
node id is all 0s in mDns advertisements, From ToT with the below commit ID

commit 5b9f9dc66e627241659f57936cbcb9455254a289 (HEAD -> master, origin/master, origin/HEAD)
Author: Zang MingJie <zealot0630@gmail.com>
Date:   Fri Apr 23 02:37:13 2021 +0800

    Increase limit of max number test suits (#6229)

[image]
[image]

franck-apple, 2021-04-22:
relates to #6216

LuDuda, 2021-04-22:
I can't reproduce it anymore after the fix was merged to the master. Can you ensure the build is clean and Thread Border Router reset before testing (in case it holds some SRP registry with already registered ID)?

franck-apple, 2021-05-17:
@cjandhyala can we close this?