project-chip/connectedhomeip #6241: Add a chip::VendorId type

bzbarsky-apple, 2021-04-22:
#### Problem
We're hardcoding `uint16_t` for vendor-id right now.

 #### Proposed Solution
Have `chip::VendorId`, make sure we use it consistently.

gerickson, 2021-05-05:
Happy to take this on; however, is this type sufficiently covered at present by https://github.com/project-chip/connectedhomeip/blob/master/src/lib/core/CHIPVendorIdentifiers.hpp#L37?

Or, would we prefer this structure:

```
#include <cstdint>

namespace chip {

//
// CHIP Vendor Identifiers (16 bits max)
//

typedef uint16_t VendorId;

enum
{
    Common       = 0x0000u,
    NestLabs     = 0x235Au,
    NotSpecified = 0xFFFFu
};

} // namespace chip
```

bzbarsky-apple, 2021-05-14:
I'm not sure why I phrased the above as "add a chip::VendorId" type, since we absolutely do have that, and I had used it.. it's possible I was filing this in parallel with a spec issue and lost track of things....

We do have a bunch of places where we use `uint16_t` for vendor ids still (`git grep "uint16_t [a-zA-Z]*[vV]endor" | grep -v gen`) that we might want to think about converting to `chip::VendorId`.

tcarmelveilleux, 2022-01-28:
This is done