project-chip/connectedhomeip #6251: missing locking

mspang, 2021-04-23:
CHIP device layer starts an IO thread to be able to respond to network activity, timers, etc. Once this task starts, calls to CHIP on any other thread must be guarded by LockChipStack/UnlockChipStack to avoid data races.

This locking is missing in many places where it is required, and must be added.

A simple approach would be to have any other threads queue their tasks via ScheduleWork() as this will invoke the task on the IO thread with the stack lock held.

See #6155, https://github.com/project-chip/connectedhomeip/pull/6207 for examples of the damage we're seeing that results from improper synchronization.

erjiaqing, 2021-04-26:
I would also suggest to discuss the return value of some APIs here.

I also agree that most APIs can use ScheduleWork() -- especially those just send messages, however, this also means that the code becomes asynchronous for those APIs, this may change the way of error handling code.

andy31415, 2021-04-28:
In python I added a 'ScheduleWorkAndWait' for the locking aspect

closed 2022-10-18 by stale[bot]