project-chip/connectedhomeip #6255: Generated emberAfFillCommandNetworkCommissioning macros seems to be broken.

kkasperczyk-no, 2021-04-23:
#### Problem
Looking at e.g. https://github.com/project-chip/connectedhomeip/blob/master/examples/lighting-app/lighting-common/gen/client-command-macro.h#L2004 and many others it seems that define should be named `emberAfFillCommandNetworkCommissioningClusterScanNetworksResponse`. It looks like some kind of formatting error during defines generation.

 #### Proposed Solution
Figure out the cause of a generation problem, fix it and regenerate files using ZAP tool.

vivien-apple, 2021-04-26:
I guess the root of the issue is https://github.com/project-chip/zap/blob/74c34f59092a9e593a4826dc657e41f85859ea55/src-electron/generator/helper-zcl.js#L238

This line takes the raw `<name>Network Commissioning</name>` field from the xml definition. Doing something like `n = n.concat(helperC.asType(el.name))` should fix the issue.

woody-apple, 2022-02-07:
This is dead code at this point.

closed 2022-02-07 by woody-apple