project-chip/connectedhomeip #6264: Wrong MAC address is used on some Linux platforms (GetPrimaryWiFiMACAddress) for mDNS

jrhees-cae, 2021-04-23:
mDNS relies on the MAC address for its advertising hostname. As such, it uses ConfigurationManagerImpl::GetPrimaryWiFiMACAddress on Linux systems (e.g., systems without Thread).

Since such systems could have:
a) multiple WiFi network interfaces
b) Ethernet network interfaces

and considering that on some platforms, the 'Primary' interface might be Ethernet, etc.

Propose:
1) Change API to something more generic, e.g. ConfigurationManagerImpl::GetPrimaryNetworkInterfaceMACAddress
2) Continue to use existing algorithm, but modified such that a CHIPConfig #define can optionally specify the target network name.

mspang, 2021-04-25:
What is a "primary" network interface?

jrhees-cae, 2021-04-27:
> What is a "primary" network interface?

IMO, a 'primary' interface would be:
a) an IP interface over which CHIP can function
b) ideally the 'most likely' interface where CHIP will be used.

Consider an example device where this is a problem: a WiFi router implementing Thread. The primary interface is NOT the WAN interface, but is likely the LAN interface. Often these devices implement WAN on eth0, and LAN on eth1, so selecting the MAC address from the first packet interface on the list is not ideal.

closed 2022-12-20 by stale[bot]