project-chip/connectedhomeip #6274: [TE2] chip-echo-responder Does not work with ipv6 addresses

jrhees-cae, 2021-04-25:
To test from a Thread device using the shell app, pinging from the Thread device to a linux machine running chip-echo-responder requires chip-echo-responder and chip-echo-requester to work with ipv6 addresses.

As an example, running chip-shell (or chip-echo-requester) on one Linux PC A, and chip-echo-responder on another Linux PC B:

1. Verify can ping from PC A to PC B via standard ipv4 ping.
2. Verify can ping from PC A to PC B via standard ipv6 ping.
3. Launch chip-shell on PC A
4. Launch chip-echo-responder on PC B
5. Test ping using chip-shell using PC B's ipv4 address
    -- success
6. Test ping using chip-shell using PC B's ipv6 address
    -- failure

Same results are obtained by using chip-echo-requester

568760310, 2021-04-29:
I found the same issue， on RP4 with TE2 image
```
4:28-21:19:44-266    ubuntu@ubuntuservercontroller:~$ chip-shell 
04:28-21:19:44-281    Init CHIP Stack
04:28-21:19:44-281    Init Thread stack
04:28-21:19:44-296    CHIP:DL: ../../src/platform/Linux/ThreadStackManagerImpl.cpp 94: Otbr ClientError 23
04:28-21:19:44-296    ThreadStackMgr().InitThreadStack() failed
04:28-21:19:44-296    > CHIP:DL: CHIP task running
04:28-21:19:51-027    ping -r 1 -c 1 10.11.97.95
04:28-21:19:51-027    IP address: 10.11.97.95
04:28-21:19:51-027    CHIP:IN: TransportMgr initialized
04:28-21:19:51-027    CHIP:IN: TransportMgr initialized
04:28-21:19:51-027    CHIP:IN: local node id is 0x000000000001b669
04:28-21:19:51-027    CHIP:IN: New pairing for device 0x0000000000bc5c01, key 0!!
04:28-21:19:51-043    Establish secure session succeeded
04:28-21:19:51-043    
04:28-21:19:51-044    Send echo request message with payload size: 32 bytes to Node: 12344321
04:28-21:19:51-044    CHIP:IN: Secure message was encrypted: Msg ID 0
04:28-21:19:51-044    CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000bc5c01 at utc time: 87832321 msec
04:28-21:19:51-044    CHIP:IN: Sending secure msg on generic transport
04:28-21:19:51-044    CHIP:IN: Secure msg send status No Error
04:28-21:19:51-045    CHIP:EM: Rxd Ack; Removing MsgId:00000000 from Retrans Table
04:28-21:19:51-045    CHIP:EM: Removed CHIP MsgId:00000000 from RetransTable
04:28-21:19:51-046    CHIP:EM: Sending Standalone Ack for MsgId:00000000
04:28-21:19:51-046    CHIP:IN: Secure message was encrypted: Msg ID 1
04:28-21:19:51-047    CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000bc5c01 at utc time: 87832322 msec
04:28-21:19:51-047    CHIP:IN: Sending secure msg on generic transport
04:28-21:19:51-047    CHIP:IN: Secure msg send status No Error
04:28-21:19:51-047    CHIP:EM: Flushed pending ack for MsgId:00000000
04:28-21:19:51-047    Echo Response: 1/1(100.00%) len=32 time=0.002ms
04:28-21:19:52-039    Done
04:28-21:19:52-825    > 
04:28-21:19:53-299    > 
04:28-21:19:56-309    > [A
04:28-21:19:56-309    Error [A: CHIP Error 4047 (0x00000FCF): Invalid argument
04:28-21:20:24-103    > ping -r 1 -c 1 fdfb:6941:8e7b:0:dea6:32ff:fef3:8725 
04:28-21:20:24-103    IP address: fdfb:6941:8e7b:0:dea6:32ff:fef3:8725
04:28-21:20:24-103    CHIP:IN: TransportMgr initialized
04:28-21:20:24-103    CHIP:IN: TransportMgr initialized
04:28-21:20:24-103    CHIP:IN: local node id is 0x000000000001b669
04:28-21:20:24-125    CHIP:IN: Connection from 'UDP:10.11.97.95:11097' expired
04:28-21:20:24-125    CHIP:IN: New pairing for device 0x0000000000bc5c01, key 0!!
04:28-21:20:24-126    Establish secure session succeeded
04:28-21:20:24-126    
04:28-21:20:24-126    Send echo request message with payload size: 32 bytes to Node: 12344321
04:28-21:20:24-126    CHIP:IN: Secure message was encrypted: Msg ID 0
04:28-21:20:24-126    CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000bc5c01 at utc time: 87865395 msec
04:28-21:20:24-126    CHIP:IN: Sending secure msg on generic transport
04:28-21:20:24-126    CHIP:IN: Secure msg send status No Error
04:28-21:20:25-115    No response received
04:28-21:20:25-115    Done
```

jrhees-cae, 2021-04-29:
Yep. I simply changed the following lines for TCP and UDP inits and rebuilt:

-            chip::Transport::TcpListenParameters(&chip::DeviceLayer::InetLayer).SetAddressType(chip::Inet::kIPAddressType_IPv4));
+            chip::Transport::TcpListenParameters(&chip::DeviceLayer::InetLayer).SetAddressType(chip::Inet::kIPAddressType_IPv6));

Of course, you could adjust such that it listens on both ipv4 and ipv6 as well.

woody-apple, 2022-01-25:
Given this is an older issue, is this complete at this point? If not, please re-open.

closed 2022-01-25 by woody-apple