project-chip/connectedhomeip #6279: failed to pairing thread device by BLE

568760310, 2021-04-26:
#### On RP4， I try to pairing a thread device by BLE,but failed. 

here is the log:
(python_env) ubuntu@ubuntu:~/te2/connectedhomeip$ chip-device-ctrl 
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x000000000001b669
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:DL: CHIP task running
CHIP:DL: Platform main loop started.
Chip Device Controller Shell

chip-device-ctrl > connect -ble 3840 20202021 1
Device is assigned with nodeid = 1
CHIP:BLE: Assigned local session key ID 0
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:EM: SessionEstablishmentExchangeDispatch::SendMessageImpl  mTransportMgr 0x25474d70
CHIP:IN: Message appended to BLE send queue
CHIP:BLE: Sent PBKDF param request
CHIP:DL: TRACE: Bus acquired for name C-105d
CHIP:DL: PlatformBlueZInit init success
CHIP:BLE: BLE removing known devices.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: Device 6F:E5:6E:BA:D7:F7 does not look like a CHIP device.
CHIP:BLE: Device 3C:22:FB:1E:CB:52 does not look like a CHIP device.
CHIP:BLE: Device 44:33:AD:AF:C1:CA does not look like a CHIP device.
CHIP:BLE: Device 59:B7:08:FF:67:3D does not look like a CHIP device.
CHIP:BLE: Device 4E:A9:67:4F:A2:D2 does not look like a CHIP device.
CHIP:BLE: Device C8:28:32:9A:30:72 does not look like a CHIP device.
CHIP:BLE: Device 4E:F5:B2:87:90:33 does not look like a CHIP device.
CHIP:BLE: Device 54:EF:44:CC:E6:6B does not look like a CHIP device.
CHIP:BLE: New device scanned: 00:60:37:29:23:3F
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Scan complete notification without an active scan.
CHIP:DL: ConnectDevice complete
 #### CHIP:DL: FAIL: NULL service in BluezConnectionInit
CHIP:DL: New BLE connection 0xffff98048a40, device 00:60:37:29:23:3F, path /org/bluez/hci0/dev_00_60_37_29_23_3F
CHIP:DL: HandlePlatformSpecificBLEEvent 16385
CHIP:IN: BleConnectionComplete: endPoint 0xffffad0452c0
 #### CHIP:DL: C1 is NULL in SendWriteRequestImpl
 #### CHIP:BLE: connect handshake timed out, closing ep 0xffffad0452c0
 #### CHIP:IN: Failed to establish BLE connection: Ble Error 6021 (0x00001785): Connect handshake timed out
CHIP:IN: Clearing BLE pending packets.
CHIP:BLE: Auto-closing end point's BLE connection.
CHIP:DL: Closing BLE GATT connection (con 0xffff98048a40)
CHIP:DL: BluezDisconnect peer=00:60:37:29:23:3F
CHIP:DL: Bluez disconnected
CHIP:DL: Bluez notify CHIPoBluez connection disconnected
Failed to establish secure session to device: 4050
CHIP Error 4050 (0x00000FD2): Timeout 

 #### code commit :2de15f4169ac2d75f905bc5fa211dcf08bc3d27b

bukepo, 2021-04-26:
I think you should set pairing credentials before `connect -ble`, for example:

```bash
set-pairing-thread-credential 16 1234 11223344556677889900aabbccddeeff
```

568760310, 2021-04-26:
Pairing Thread Credential is nolonger available.  To use NetworkCommissioning cluster,  thread device must be connected with BLE firstly.

568760310, 2021-04-26:
Pairing Thread Credential is nolonger available. To use NetworkCommissioning cluster, thread device must be connected with BLE firstly.

> I think you should set pairing credentials before `connect -ble`, for example:
> 
> ```shell
> set-pairing-thread-credential 16 1234 11223344556677889900aabbccddeeff
> ```

568760310, 2021-04-27:
More test  using RPi 4 with [TE2 image ](https://drive.google.com/file/d/1-q9_84xKxBgH38uZ3l-Ll9b5RmlnaZyy/view?usp=sharing)and K32W board.

####  Harness Logs
```
ubuntu@ubuntuservercontroller:~$ chip-device-ctrl 
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x000000000001b669
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:DL: CHIP task running
CHIP:DL: Platform main loop started.
Chip Device Controller Shell

chip-device-ctrl > connect -ble 3840 20202021 1
Device is assigned with nodeid = 1
CHIP:BLE: Assigned local session key ID 0
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:DL: TRACE: Bus acquired for name C-0c19
CHIP:EM: SessionEstablishmentExchangeDispatch::SendMessageImpl  mTransportMgr 0x153ecd70
CHIP:IN: Message appended to BLE send queue
CHIP:BLE: Sent PBKDF param request
CHIP:DL: PlatformBlueZInit init success
CHIP:BLE: BLE removing known devices.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: New device scanned: 00:60:37:29:23:3F
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Scan complete notification without an active scan.
CHIP:DL: ConnectDevice complete
CHIP:DL: FAIL: NULL service in BluezConnectionInit
CHIP:DL: New BLE connection 0xffffa80419e0, device 00:60:37:29:23:3F, path /org/bluez/hci0/dev_00_60_37_29_23_3F
CHIP:DL: HandlePlatformSpecificBLEEvent 16385
CHIP:IN: BleConnectionComplete: endPoint 0xffffb355e1a0
CHIP:DL: C1 is NULL in SendWriteRequestImpl
CHIP:BLE: connect handshake timed out, closing ep 0xffffb355e1a0
CHIP:IN: Failed to establish BLE connection: Ble Error 6021 (0x00001785): Connect handshake timed out
CHIP:IN: Clearing BLE pending packets.
CHIP:BLE: Auto-closing end point's BLE connection.
CHIP:DL: Closing BLE GATT connection (con 0xffffa80419e0)
CHIP:DL: BluezDisconnect peer=00:60:37:29:23:3F
CHIP:DL: Bluez disconnected
CHIP:DL: Bluez notify CHIPoBluez connection disconnected
Failed to establish secure session to device: 4050
CHIP Error 4050 (0x00000FD2): Timeout
chip-device-ctrl > 
```

#### Device Logs
```
[Info]Assigned local session key ID 0
[Info]Waiting for PBKDF param request
[Info]Started BLE Advertising!
[Info]Started Advertising.
[Info]Thread joiner timer running
[Info]Request Thread joiner start
[Info]Joiner Discerner: 3840
[Info]Joiner PSKd: 020202021
[Info]Joiner start: No Error
[Info]OpenThread State Changed (Flags: 0x0800800b)
[Info]   Network Name: OpenThread
[Info]   PAN Id: 0x2DE9
[Info]   Extended PAN Id: 0xDEAD00BEEF00CAFE
[Info]   Channel: 11
[Info]   Mesh Prefix: FDDE:AD00:BEEF::/64
[Info]   Master Key: A4CFE5B3A2B8E713AB1988D991BA13E9
[Info]   Thread Unicast Addresses:
[Info]        FE80::289F:9FE:D9F6:254F/64 valid preferred
[Info]LwIP Thread interface addresses updated
[Info]   FE80::289F:9FE:D9F6:254F IPv6 link-local address, preferred)
[Info]BLE is connected with device: 0.

[Info]Started Advertising.
[Info]BLE MTU size has been changed to 247.
[Info]Join Thread network: NotFound
[Info]OpenThread State Changed (Flags: 0x0800800b)
[Info]   Network Name: OpenThread
[Info]   PAN Id: 0xFFFF
[Info]   Extended PAN Id: 0xDEAD00BEEF00CAFE
[Info]   Channel: 11
[Info]   Mesh Prefix: FDDE:AD00:BEEF::/64
[Info]   Master Key: A4CFE5B3A2B8E713AB1988D991BA13E9
[Info]   Thread Unicast Addresses:
[Info]        FE80::40D4:1BD0:4293:CF00/64 valid preferred
[Info]LwIP Thread interface addresses updated
[Info]   FE80::40D4:1BD0:4293:CF00 IPv6 link-local address, preferred)
[Info]Thread joiner timer running
[Info]Request Thread joiner start
[Info]Joiner Discerner: 3840
[Info]Joiner PSKd: 020202021
[Info]Joiner start: No Error
[Info]OpenThread State Changed (Flags: 0x0800800b)
[Info]   Network Name: OpenThread
[Info]   PAN Id: 0x339D
[Info]   Extended PAN Id: 0xDEAD00BEEF00CAFE
[Info]   Channel: 11
[Info]   Mesh Prefix: FDDE:AD00:BEEF::/64
[Info]   Master Key: A4CFE5B3A2B8E713AB1988D991BA13E9
[Info]   Thread Unicast Addresses:
[Info]        FE80::AC2E:C714:F3F5:5F55/64 valid preferred
[Info]LwIP Thread interface addresses updated
[Info]   FE80::AC2E:C714:F3F5:5F55 IPv6 link-local address, preferred)
[Info]Join Thread network: NotFound
[Info]OpenThread State Changed (Flags: 0x0800800b)
[Info]   Network Name: OpenThread
[Info]   PAN Id: 0xFFFF
[Info]   Extended PAN Id: 0xDEAD00BEEF00CAFE
[Info]   Channel: 11
[Info]   Mesh Prefix: FDDE:AD00:BEEF::/64
[Info]   Master Key: A4CFE5B3A2B8E713AB1988D991BA13E9
[Info]   Thread Unicast Addresses:
[Info]        FE80::ACC7:8209:5087:F00/64 valid preferred
[Info]LwIP Thread interface addresses updated
[Info]   FE80::ACC7:8209:5087:F00 IPv6 link-local address, preferred)
[Info]Thread joiner timer running
[Info]Request Thread joiner start
[Info]Joiner Discerner: 3840
[Info]Joiner PSKd: 020202021
[Info]Joiner start: No Error
[Info]OpenThread State Changed (Flags: 0x0800800b)
[Info]   Network Name: OpenThread
[Info]   PAN Id: 0xBB53
[Info]   Extended PAN Id: 0xDEAD00BEEF00CAFE
[Info]   Channel: 11
[Info]   Mesh Prefix: FDDE:AD00:BEEF::/64
[Info]   Master Key: A4CFE5B3A2B8E713AB1988D991BA13E9
[Info]   Thread Unicast Addresses:
[Info]        FE80::18F9:6654:C3C8:C59C/64 valid preferred
[Info]LwIP Thread interface addresses updated
[Info]   FE80::18F9:6654:C3C8:C59C IPv6 link-local address, preferred)
[Info]Join Thread network: NotFound
[Info]OpenThread State Changed (Flags: 0x0800800b)
[Info]   Network Name: OpenThread
[Info]   PAN Id: 0xFFFF
[Info]   Extended PAN Id: 0xDEAD00BEEF00CAFE
[Info]   Channel: 11
[Info]   Mesh Prefix: FDDE:AD00:BEEF::/64
[Info]   Master Key: A4CFE5B3A2B8E713AB1988D991BA13E9
[Info]   Thread Unicast Addresses:
[Info]        FE80::B8F5:394C:D4E2:8F00/64 valid preferred
[Info]LwIP Thread interface addresses updated
[Info]   FE80::B8F5:394C:D4E2:8F00 IPv6 link-local address, preferred)
[Info]Stopped connectionTimeout timer.
[Info]BLE is disconnected with device: 0.
[Info]Advertising error!
[Info]Disabling CHIPoBLE service due to error: CHIP Error 4000003 (0x003D0903): Incorrect state
```

568760310, 2021-04-28:
@doru91  
Thanks ， according you suggestions:

 "replace PRIMARY_SERVICE(service_chipoble, gChipoBleService_d) with PRIMARY_SERVICE(service_chipoble, 0xFFF6)  inside src/platform/K32W/gatt_db.h . "
I fixed this bug .

closed 2021-04-28 by 568760310