project-chip/connectedhomeip #6306: [TE2] chip-device-ctrl can't do NetworkCommissioning

wissn84, 2021-04-27:
#### Problem
- controller (chip-device-ctrl) in linux
- controllee (chip-lighting-app) in linux 
- branch : test_event_2.2

I have tried to Commissioning between chip-device-ctrl and chip-lighting-app 

1. chip-device-ctrl > connect -ble 3840 20202021

`Secure Session to Device Established
Device temporary node id (**this does not match spec**): 860392
chip-device-ctrl > CHIP:DL: HandlePlatformSpecificBLEEvent 16387`

2. zcl NetworkCommissioning AddWiFiNetwork 12344321 1 0 ssid=str:chip credentials=str:qwerasdf breadcrumb=0 timeoutMs=1000

then I got error message like below
`An exception occurred during process ZCL command:
CHIP Error 4072 (0x00000FE8): Not connected`

So I check log of chip-lighting-app and got error called "Failed to store ~~"

`CHIP:SVR: Device completed SPAKE2+ handshake
CHIP:SVR: Device completed Rendezvous process
CHIP:SVR: Failed to store the connection state
CHIP:DL: UnregisterAdvertisement complete`

There exist Persistent Storage in in tmp folder but  There is empty file

Log is attached

 #### Proposed Solution
<suggested fix, suggested enhancement>

erjiaqing, 2021-04-27:
The nodeid is `860392` not `12344321`

woody-apple, 2022-01-25:
Given this is an older issue, is this complete at this point? If not, please re-open.

closed 2022-01-25 by woody-apple