project-chip/connectedhomeip #6311: BLE adapter assignment not happening as expected with -b, it pics the virtual interfaces by default unless i overwrite it using ble-adapter-select

SaileeG, 2021-04-27:
If I pass -b hci0 to the chip device controller I am able to scan for only virtual inetrface(Linux Lighting app)

**Expected Behavior:**
When I run the chip-device-ctroller -b hci0 and initiate a ble scan it should scan the Physical dev boards or accessories.

**Behavior seen:**
```
root@f366b03329bc:/connectedhomeip# chip-device-ctrl -b hci0
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 112233

CHIP:DL: CHIP task running
Chip Device Controller Shell

chip-device-ctrl > ble-scan
2021-04-26 09:19:38,061 ChipBLEMgr   INFO     use default adapter
2021-04-26 09:19:38,953 ChipBLEMgr   INFO     scanning started
2021-04-26 09:19:39,177 ChipBLEMgr   INFO     Name            = CHIP-3840
2021-04-26 09:19:39,177 ChipBLEMgr   INFO     ID              = 2193682d-4caf-3a4a-9713-03a1a3329db8
2021-04-26 09:19:39,179 ChipBLEMgr   INFO     RSSI            = 127
2021-04-26 09:19:39,181 ChipBLEMgr   INFO     Address         = 00:AA:01:00:00:23
2021-04-26 09:19:39,187 ChipBLEMgr   INFO     Pairing State   = 0
2021-04-26 09:19:39,187 ChipBLEMgr   INFO     Discriminator   = 3840
2021-04-26 09:19:39,187 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-04-26 09:19:39,188 ChipBLEMgr   INFO     Product Id      = 65279
2021-04-26 09:19:39,192 ChipBLEMgr   INFO     Adv UUID        = 0000feaf-0000-1000-8000-00805f9b34fb
2021-04-26 09:19:39,194 ChipBLEMgr   INFO     Adv Data        = 00000f5a23fffe
2021-04-26 09:19:39,194 ChipBLEMgr   INFO     
2021-04-26 09:19:49,262 ChipBLEMgr   INFO     scanning stopped

```
We need to again expicity delect the hci0 using ble-adapter-select only then we are able to scan for the physical boards
```
chip-device-ctrl > ble-adapter-select hci0
This change only applies to ble-scan
Please run device controller with --bluetooth-adapter=<adapter-name> to select adapter
e.g. chip-device-ctrl --bluetooth-adapter hci0
chip-device-ctrl > ble-scan
2021-04-26 09:20:02,092 ChipBLEMgr   INFO     scanning started
2021-04-26 09:20:02,392 ChipBLEMgr   INFO     Name            = ChipLight
2021-04-26 09:20:02,392 ChipBLEMgr   INFO     ID              = f016e23d-0d00-35d5-93e7-588acdbc7e54
2021-04-26 09:20:02,394 ChipBLEMgr   INFO     RSSI            = -37
2021-04-26 09:20:02,396 ChipBLEMgr   INFO     Address         = D4:BC:D3:69:E5:DC
2021-04-26 09:20:02,402 ChipBLEMgr   INFO     Pairing State   = 0
2021-04-26 09:20:02,402 ChipBLEMgr   INFO     Discriminator   = 3840
2021-04-26 09:20:02,402 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-04-26 09:20:02,402 ChipBLEMgr   INFO     Product Id      = 65279
2021-04-26 09:20:02,406 ChipBLEMgr   INFO     Adv UUID        = 0000feaf-0000-1000-8000-00805f9b34fb
2021-04-26 09:20:02,408 ChipBLEMgr   INFO     Adv Data        = 00000f5a23fffe
2021-04-26 09:20:02,409 ChipBLEMgr   INFO     
2021-04-26 09:20:12,222 ChipBLEMgr   INFO     scanning stopped

chip-device-ctrl > ble-adapter-select hci2
This change only applies to ble-scan
Please run device controller with --bluetooth-adapter=<adapter-name> to select adapter
e.g. chip-device-ctrl --bluetooth-adapter hci0
chip-device-ctrl > ble-scan
2021-04-26 09:20:17,868 ChipBLEMgr   INFO     scanning started
2021-04-26 09:20:18,176 ChipBLEMgr   INFO     Name            = CHIP-3840
2021-04-26 09:20:18,176 ChipBLEMgr   INFO     ID              = 2193682d-4caf-3a4a-9713-03a1a3329db8
2021-04-26 09:20:18,178 ChipBLEMgr   INFO     RSSI            = 127
2021-04-26 09:20:18,180 ChipBLEMgr   INFO     Address         = 00:AA:01:00:00:23
2021-04-26 09:20:18,185 ChipBLEMgr   INFO     Pairing State   = 0
2021-04-26 09:20:18,186 ChipBLEMgr   INFO     Discriminator   = 3840
2021-04-26 09:20:18,186 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-04-26 09:20:18,186 ChipBLEMgr   INFO     Product Id      = 65279
2021-04-26 09:20:18,190 ChipBLEMgr   INFO     Adv UUID        = 0000feaf-0000-1000-8000-00805f9b34fb
2021-04-26 09:20:18,192 ChipBLEMgr   INFO     Adv Data        = 00000f5a23fffe
2021-04-26 09:20:18,192 ChipBLEMgr   INFO     
2021-04-26 09:20:28,230 ChipBLEMgr   INFO     scanning stopped
chip-device-ctrl >

```

erjiaqing, 2021-04-29:
This issue should be fixed in https://github.com/project-chip/connectedhomeip/pull/5805 already, and the command output indicates that this is an outdated checkout.
The chip-device-ctrl should print

```
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x000000000001b669
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:DL: CHIP task running
CHIP:DL: Platform main loop started.
Chip Device Controller Shell

chip-device-ctrl > 
```

on start.

SaileeG, 2021-04-30:
Works with Latest Master

closed 2021-04-30 by SaileeG