project-chip/connectedhomeip #6312: OTBR used for TE2 is not answering to Dynamic Update Queries

doru91, 2021-04-27:
I am trying to test NXP ELock Demo application with mDNS. The problem is that the OTBR does not reply to the Dynamic Update Request initiated by the SRP client on the Thread device.

There is also a strange UDP Checksum Error in the OTBR Logs but not sure if this is relevant.

Also, please see attached the Wireshark capture (unzip it first).

[image]
[image]

[no_dynamic_update_response_wireshark.zip](https://github.com/project-chip/connectedhomeip/files/6384866/no_dynamic_update_response_wireshark.zip)

doru91, 2021-04-27:
I've also analyzed the otbr logs: it seems that there are some mDNS logs but these are related to IPv4 (so maybe Ethernet traffic). So mDNS is enabled on OTBR. I would expected to see mDNS logs also for IPv6 (Thread) traffic but I don't see. What I see is those UDP Checksum errors: so maybe the mDNS queries originated from the Thread device are dropped due to UDP checksum errors?

LuDuda, 2021-04-27:
It might be something else, but please verify that you provisioned a full Thread Dataset to the device. The `Error UDP Checksum` appears usually when you have a different Mesh Local Prefix on the Border Router and on the Thread device.

Explanation: The Mesh Local Prefix is compressed on 6LoWPAN layer with Context ID 0. Because of that the prefix is not sent over the air, and devices need to have the same address to properly calculate the checksum. Note that UDP Checksum for IPv6 includes both source and destination address.

doru91, 2021-04-28:
Thanks, Lukasz, that was the problem: a different ML prefix.

closed 2021-04-28 by doru91