project-chip/connectedhomeip #6329: [TE2] resolve command is not response in chip-device-ctrl

wissn84, 2021-04-28:
#### Problem
After close-ble then resolve command is not response in chip-device-ctrl.

Controller : chip-device-ctrl / ubuntu
Controllee  : chip-lighting-app / ubuntu
Version : TE2

avahi result is below
` avahi-browse -rt _chip._tcp
+   eth0 IPv6 0000000000000000-0000000000BC5C01             _chip._tcp           local
+   eth0 IPv4 0000000000000000-0000000000BC5C01             _chip._tcp           local
+  wlan0 IPv6 0000000000000000-0000000000BC5C01             _chip._tcp           local
=   eth0 IPv6 0000000000000000-0000000000BC5C01             _chip._tcp           local
   hostname = [DCA632D839F9.local]
   address = [192.168.1.3]
   port = [11097]
   txt = ["="]
=   eth0 IPv4 0000000000000000-0000000000BC5C01             _chip._tcp           local
   hostname = [DCA632D839F9.local]
   address = [192.168.1.3]
   port = [11097]
   txt = ["="]
=  wlan0 IPv6 0000000000000000-0000000000BC5C01             _chip._tcp           local
   hostname = [DCA632D839F9.local]
   address = [192.168.1.3]
   port = [11097]
   txt = ["="]`

below is resolve command after no response
`chip-device-ctrl > CHIP:DL: BluezDisconnect peer=00:1A:7D:DA:71:03
CHIP:DL: HandlePlatformSpecificBLEEvent 16388
CHIP:BLE: no endpoint for unsub complete
CHIP:DL: Bluez disconnected
CHIP:DL: Bluez notify CHIPoBluez connection disconnected
resolve 0 12344321
`

Log is attached.

 #### Proposed Solution

doru91, 2021-04-28:
I am also experiencing the same issue. I tried to look into the chip-device-ctrl code and it seems that when the resolve command is issued then an IPV6 broadcast packet should be initiated by the chip-device-ctrl for resolving the FabricID/Node Id. Then I tried to use tcpdump on otbr0 interface but no packet is captured.

tlykkeberg-grundfos, 2021-04-28:
We are also experiencing this, but with device hardware from both TI and Nordic in the lock-app example. We are also using the chip-device-ctrl in Ubuntu (on the OTBR).

```
ubuntu@ubuntu:~/connectedhomeip-te2$ avahi-browse -tr '_chip._tcp'
+  otbr0 IPv6 000000014A77CBB3-0000000000BC5C01             _chip._tcp           local
+  otbr0 IPv4 000000014A77CBB3-0000000000BC5C01             _chip._tcp           local
=  otbr0 IPv4 000000014A77CBB3-0000000000BC5C01             _chip._tcp           local
   hostname = [F4CE36F4F037DBB0.local]
   address = [fd11:22::3bc0:af8d:9bda:1dfb]
   port = [11097]
   txt = ["��\000=300" "��\000=5000"]
=  otbr0 IPv6 000000014A77CBB3-0000000000BC5C01             _chip._tcp           local
   hostname = [F4CE36F4F037DBB0.local]
   address = [fd11:22::3bc0:af8d:9bda:1dfb]
   port = [11097]
   txt = ["��\000=300" "��\000=5000"]
``` 
And the `chip-device-ctrl > resolve 5544332211 12344321` does not respond at all, it just keeps spinning.

But ping'ing the device from the Ubuntu (OTBR) works fine.

```
ubuntu@ubuntu:~/connectedhomeip-te2$ ping6 fd11:22::3bc0:af8d:9bda:1dfb
PING fd11:22::3bc0:af8d:9bda:1dfb(fd11:22::3bc0:af8d:9bda:1dfb) 56 data bytes
64 bytes from fd11:22::3bc0:af8d:9bda:1dfb: icmp_seq=1 ttl=63 time=17.8 ms
64 bytes from fd11:22::3bc0:af8d:9bda:1dfb: icmp_seq=2 ttl=63 time=17.7 ms
```

Damian-Nordic, 2021-04-28:
Note that by default on Linux platform the in-house minimal mDNS implementation is used which apparently has some issues. I suggest that you build the Python CHIP controller with `chip_mdns="platform"` GN arg so that Avahi-based client is used.

tlykkeberg-grundfos, 2021-04-28:
Thanks @Damian-Nordic that made it work for the Nordic hardware 👍 But the TI is still not working. Perhaps it does not use mDNS as I expect?

LuDuda, 2021-04-28:
I don't know TI platform, but apparently they don't have SRP enabled. You can experiment with setting `chip_mdns = platform` like EFR [did](https://github.com/project-chip/connectedhomeip/blob/14e0ee241759994614a22678ec573db4f6408a8a/src/platform/EFR32/args.gni#L28) lately.

doru91, 2021-04-28:
Thanks, Damien. This also fixed the error on my side.

wissn84, 2021-04-29:
Thanks, It is Working well in chip-lighting-app.

closed 2021-04-29 by wissn84