project-chip/connectedhomeip #6396: Pairing with ble using chip-tool fails.

sweetymhaiske, 2021-04-30:
#### Problem
Using chip-tool, pairing with `ble-wifi` fails as after successful response from device to chip-tool, device keeps on sending notify events to chip-tool and auto-close BLE connection.

**Device Logs**
```
GATT procedure initiated: notify; att_handle=14
I (286376) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
I (289136) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
GATT procedure initiated: notify; att_handle=14
I (291646) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
I (294406) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
GATT procedure initiated: notify; att_handle=14
I (296906) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
I (297556) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 0 ) indicate = 0 notify = 0
I (297556) chip[DL]: CHIPoBLE unsubscribe received
V (297556) chip[IN]: Clearing BLE pending packets.
I (297566) chip[BLE]: Releasing end point's BLE connection back to application.
I (299446) chip[DL]: BLE GAP connection terminated (con 37910
```

**Chip-tool Logs**
```
CHIP:DL: Indication received, conn = 0x7f3070033060
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0x7f3070033060
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0x7f3070033060
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0x7f3070033060
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0x7f3070033060
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:TOO: No response from device
CHIP:CTL: Shutting down the commissioner
CHIP:CTL: Shutting down the controller
CHIP:DL: System Layer shutdown
CHIP:DL: Inet Layer shutdown
CHIP:DL: BLE layer shutdown
CHIP:IN: Clearing BLE pending packets.
CHIP:BLE: Auto-closing end point's BLE connection.
CHIP:DL: Closing BLE GATT connection (con 0x7f3070033060)
CHIP:DL: BluezDisconnect peer=7D:13:E3:5B:4E:9A

```

nageshshamnur, 2021-05-18:
@sweetymhaiske: for ./chip-tool pairing ble-wifi <ssid> <password> <fabric-id>

can you tell me how did you arrive at fabric-id value, i am unable to figure it out from any documentation. Thanks

dhrishi, 2021-09-09:
@sweetymhaiske Please close this if it is not valid now

sweetymhaiske, 2021-09-09:
Resolved.

closed 2021-09-09 by sweetymhaiske