project-chip/connectedhomeip #6421: Query regarding building CHIP for MCUs which are supported by zephyr

nageshshamnur, 2021-05-03:
#### Problem
There is a limited documentation on how to build CHIP for MCUs. I see under the documentation https://github.com/project-chip/connectedhomeip/tree/master/docs/guides, it's explained for limited architectures and boards. 

So, all the boards which zephyr supports (https://docs.zephyrproject.org/2.5.0/boards/index.html), is currently supported or not? if supported, then kindly guide me to relevant documentation on how to build the same.

Many thanks in advance.

 #### Proposed Solution
Documentation pages explaining on how to build CHIP for zephyr OS.

nageshshamnur, 2021-05-03:
Also i see that Zephyr is supported via nRF Connect SDK which supports only limited set of boards(https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/ug_nrf52.html#supported-boards) and doesn't support all the boards which is supports by zephyr(https://docs.zephyrproject.org/2.5.0/boards/index.html).

nageshshamnur, 2021-05-04:
@Damian-Nordic

Damian-Nordic, 2021-05-05:
@nageshshamnur As you noticed, Zephyr is currently supported only via nRF Connect SDK. We tried to extract common Zephyr code to `src/platform/Zephyr`, so that it can be easily reused by other Zephyr platforms, but so far we have only added a full support for nRF Connect SDK.

nageshshamnur, 2021-05-05:
> @nageshshamnur As you noticed, Zephyr is currently supported only via nRF Connect SDK. We tried to extract common Zephyr code to `src/platform/Zephyr`, so that it can be easily reused by other Zephyr platforms, but so far we have only added a full support for nRF Connect SDK.

Thanks @Damian-Nordic for the reply. Your reply validates my understanding and answers my question as well!!!.

closed 2021-05-05 by nageshshamnur