project-chip/connectedhomeip #6456: Sort out interaction of exchange lifetime and CRMP in the SDK

bzbarsky-apple, 2021-05-04:
#### Problem
Consider a simple message flow:
1. A sends an unsolicited message to B
2. B sends a response to A, does not expect any more messages

At what point does B call `Close` on the `ExchangeContext` that it created to handle the unsolicited message?  My expectation as an API consumer would be "right after calling SendMessage" with the response.  But if we do that, CRMP will fail to retransmit that message, because CRMP retransmit needs state from the exchange context's delegate, but `Close` nulls the delegate out.

If we don't null the delegate out during `Close` but just use it when we need to retransmit, it's not clear how a consumer is supposed to know when it's safe to deallocate the delegate.

We could have some sort of delegate callback that indicates "All message sends have completed (do not need ack, got ack, or timed out waiting for ack)" or something like that, which would then allow the consumer to close the exchange.  But this is slightly annoying to use and exposes transport-level state to the higher-level API...

 #### Proposed Solution
Don't have a good one yet, see above.  Ideally we would not need anything caller-allocated to do CRMP retransmits and could just `Close` right after `SendMessage` and move on, but that's not the setup we seem to have.

@pan-apple @tcarmelveilleux @yufengwangca @erjiaqing @kghost @mrjerryjohns @vivien-apple

yufengwangca, 2021-05-04:
Each protocol needs to define the end of each conversation for both client and server side. The exchange delegate needs to call 'Close' after it sends the last message defined in the conversation.

Take Echo protocol for example, the Echo protocol define Echo Response message as the end of echo conversation, the Echo Client  should call 'Close' after it receives the Echo response and Echo Server should call 'Close' after it sends Echo response message out.

The exchange delegate can only close the exchange, which not release the exchange, but handle it over to ExchangeMgr to deal with the remaining CRMP ack if there is any. The ExchangeMgr should finally release the exchanges which have been closed after CRMP complete.

yufengwangca, 2021-05-04:
I don't think we should figure out any smart way to release the exchange automatically, since it is depending on the syntax of each conversation, only protocol has the knowledge when the conversation should be ended.

kghost, 2021-05-05:
I feel that it is nessasary to have a smart ptr like object to help managing the lifecycle of ExchangeContext. The current interface of managing the context is not well designed or documented. I'll try to work out a patch to see if we can improve it.

bzbarsky-apple, 2021-05-05:
> The exchange delegate needs to call 'Close' after it sends the last message defined in the conversation.

That's fine.  My point is that right now if that is done that nulls out the delegate on the exchange, which means the exchange is no longer able to send CRMP retransmits after that point.

> but handle it over to ExchangeMgr to deal with the remaining CRMP ack if there is any.

This is the part that is broken.  If the ack does not arrive, the CRMP bits try to retransmit and that fails because the exchange has no delegate.

> I don't think we should figure out any smart way to release the exchange automatically

I am not asking for one.  I am asking for us to have some way, any way, for a basic "I get a message, I send a response, I am done" flow work.  Right now it does not work correctly.

Note that a smart ptr for managing the exchange lifecycle, while it might be a good idea, will NOT solve the problem this issue is about, as far as I can tell.

bzbarsky-apple, 2021-05-05:
To be very clear: code in `ReliableMessageMgr::SendFromRetransTable` looks like this:
```
    const ExchangeMessageDispatch * dispatcher = rc->GetExchangeContext()->GetMessageDispatch();
    VerifyOrExit(dispatcher != nullptr, err = CHIP_ERROR_INCORRECT_STATE);    err =
        dispatcher->ResendMessage(rc->GetExchangeContext()->GetSecureSession(), std::move(entry->retainedBuf), &entry->retainedBuf);
```
but after calling `Close()` on the `ExchangeContext`, `rc->GetExchangeContext()->GetMessageDispatch()` returns null, so we error out with `CHIP_ERROR_INCORRECT_STATE` and the `ResendMessage` line is never reached.

mrjerryjohns, 2021-05-05:
IMHO part of the problem here is that we treat the CRMP messages as somehow being part of the exchange, when in reality, it shouldn't be. CRMP _should be_ a 'layer lower' than Exchange, should reliably re-transmit a message without needing any Exchange linkages. If you view CRMP as a counterpart to TCP, then this makes more sense.

Upon `SendMessage`, the protocol managing that exchange should call `Close` on the EC, which should terminate the exchange. The message at this point should be sitting in the retrans table regardless of the state of the EC. This I think, achieves better decoupling of ECs from the lifetime of the message itself.

```
const ExchangeMessageDispatch * dispatcher = rc->GetExchangeContext()->GetMessageDispatch();
```

The above layering has ExchangeContext too heavily coupled with CRMP as the transport delivery construct. This doesn't feel right, since Exchanges should be able to run over TCP transports as well.

The above should be re-factored such that dispatching a message on a given EC is handled by the message layer given some selection of transport, and the appropriate means to deliver that reliably.

bzbarsky-apple, 2021-05-05:
> The message at this point should be sitting in the retrans table regardless of the state of the EC.

Yes, this happens.

> The above layering has ExchangeContext too heavily coupled with CRMP as the transport delivery construct

I'm not sure what you mean here.  That code is the CRMP state machine trying to resend the message, but it doesn't know _how_ to do it short of asking the exchange.  TCP doesn't really enter the picture here at all.

mrjerryjohns, 2021-05-05:
> but it doesn't know how to do it short of asking the exchange

Why is CRMP asking the exchange to deliver the message? That's going ... upwards instead of downwards lower into the stack to deliver messages...

bzbarsky-apple, 2021-05-05:
> Why is CRMP asking the exchange to deliver the message? 

It's not.  It's asking it for the object that knows how to deliver messages....

yufengwangca, 2021-05-05:
"That's fine. My point is that right now if that is done that nulls out the delegate on the exchange, which means the exchange is no longer able to send CRMP retransmits after that point."

Why exchange needs a delegate to be able to do CRMP retransmits? The CRMP should still works after the protocol close the exchange and nlls out the delegate.

" 
 const ExchangeMessageDispatch * dispatcher = rc->GetExchangeContext()->GetMessageDispatch();
    VerifyOrExit(dispatcher != nullptr, err = CHIP_ERROR_INCORRECT_STATE);    err =
        dispatcher->ResendMessage(rc->GetExchangeContext()->GetSecureSession(), std::move(entry->retainedBuf), &entry->retainedBuf);
 "
 This logic is recently added for moving PASE over exchange, it seems has potential issue since GetExchangeContext() could be nullified.

turon, 2021-05-05:
It seems there are two possible solutions here:

1) Allow CRMP to effectively hold a reference count to the exchange and keep it open even though the upper layer closed it.

2) Or put the onus on the app, and encourage a programming model where  the app closes exchanges on some finalizing crmp event (ack Rx or error/timeout) and respect and truly close exchanges if app cuts it off short (before actual acks arrive).

bzbarsky-apple, 2021-05-06:
@pan-apple and I talked about this a bit, and we feel that we probably want to allow some sort of notification to the sender of a message if CRMP delivery "fails" (i.e. we run out of retransmits and don't get an ack).  Similar for TCP delivery failing (i.e. never get an ack for those bytes, eventually decide TCP connection is dead).  In both cases the message _might_ have been delivered, but we don't know for sure, and the higher layers may want to know that.

If we want to allow that, that means an exchange needs to be able to notify _someone_ once it's all done with the messages it was asked to send and whether they succeeded/failed.  That "someone" (e.g. the exchange delegate) would need to stay alive until such notification.  The exchange would guarantee that it will not touch the "someone" after that "I am done" notification.

This suggests that `Close` should have the following semantics:
* I will send no more messages, so flush out any pending acks; nothing to piggyback them on.
* Treat this exchange as no longer existing in terms of messages from the peer.  If something else comes in with I flag and this same exchange ID, that's a new exchange. (?)
* The exchange context notifies the delegate when it's done with it so the delegate can deallocate or whatever.  Notably, Close() does NOT necessarily null out the delegate.  Instead it communicates (as its return value, say) whether there is anything waiting on acks right now.  And if there is, it promises a callback to the delegate later when everything has cleared.  This is not CRMP-specific, really; TCP or BTP would have similar "I sent this, but waiting on an ack" state that it could know about.

Thoughts?

kghost, 2021-05-06:
> @pan-apple and I talked about this a bit, and we feel that we probably want to allow some sort of notification to the sender of a message if CRMP delivery "fails" (i.e. we run out of retransmits and [...]

I'd agree that `Close` means to initiate the graceful shutdown procedure, not release the context. But I'm not sure if it is good to report that a message have been acked. Because the schematic will be diverge from TCP, because TCP doesn't report that a packet has been successfully received by the peer.

> Allow CRMP to effectively hold a reference count to the exchange and keep it open even though the upper layer closed it.

This solution will provides the best interface for usability. But there is a bug that CRMP depens on ExchangeDelegate to send a message, the issue is going to fix this.

I'm working on a quick fix of this issue, where I'll decouples `MessageDispatcher` from `ExchangeDelegate`. Current we have 2 dispatchers, one for secure messages named `ApplicationExchangeDispatch`, one for unsecure messages `SessionEstablishmentExchangeDispatch`. I'm not sure if there are any further reason that Pankaj name them this way, but I want to rename them to `SecureExchangeDispatch` and `UnsecureExchangeDispatch`, and choose them statically, via a flag in ExchangeContext. (@pan-apple any suggestions ?)

> IMHO part of the problem here is that we treat the CRMP messages as somehow being part of the exchange, when in reality, it shouldn't be. CRMP should be a 'layer lower' than Exchange, should [...]

CRMP needs re-works. Lots of dead code and malformed code in the CRMP manager, for example, when message counter identifies a duplicated message, a duplicated ack should be send, but currently the duplicated message is silently dropped. I'll create another issue to track the re-work progress of CRMP.

yufengwangca, 2022-02-02:
This ticket is pretty old, and a lot of things happen since then. we need to check if this issue still exists sine exchange can be auto-released after usage now

bzbarsky-apple, 2022-02-02:
Now that the message dispatch is gotten only once and is guaranteed (by contract) to have long-enough lifetime, we can send things after Close().  So I _think_ this is fixed.

@msandstedt do you agree?

commit 0f0da383 (2021-05-13): Remove ExchangeMessageDispatch.h (Fix #6456)

Major changes:
 * ExchangeManager and ExchangeContext will handle both secure and unsecure messages
 * ExchangeContext will be explicitly flagged as secure or unsecure
    * Secure context will only handle secure messages
    * Unsecure context will only handle Unsecure messages
 * Add a simple Variant (src/lib/support/Variant.h)
    * Secure and unsecure context have diffent member fields. Use Variant to manage these fields.
 * Use RAII for ExchangeContext contruction and destruction.
    * Use contructor and destructor of ExchangeContext instread [... cut]
closed 2022-02-07 by woody-apple