project-chip/connectedhomeip #6475: Factory reset doesn't clear KVS admin pairings

cecille, 2021-05-04:
#### Problem
Factory reset doesn't reset admin pairings. This definitely happens on M5, unsure about others

 #### Proposed Solution
Reset admins on Factory reset - should be done in a way that this works for all platforms.

holbrookt, 2022-02-01:
@cecille any idea if this is still an issue?

cecille, 2022-02-01:
I'm not sure - sorry. I do think we need factory resets to be working properly for v1.0, but I'm not sure if it's something that's explicitly covered by the SDK vs. something that's covered by platform. @harimau-qirex may have a better idea as it touches on storage.

woody-apple, 2022-02-07:
Factory reset is handled per platform, as the matter doesn't really support Factory reset at this point.

closed 2022-10-09 by stale[bot]