project-chip/connectedhomeip #6481: EFR32 lock example pairing/connection  is broken on current master

saurabhst, 2021-05-05:
#### Problem
The current master has the following 2 issues wrt to EFR32 Lock app example -

1) QR code string is wrong and does not match corresponding to the PIN and discriminator .

Expected Data should be 
```setup-payload parse-qr CH:H34UKNY00VN2V0E4
Version: 0
VendorID: 9050
ProductID: 21323
RequiresCustomFlow: 0
RendezvousInformation: 2 [BLE]
Discriminator: 3840
SetUpPINCode: 73141520```

But when I scan the QR code using a QR code scanner I get the string as `CH:W0GU245600C0NF0ET00` because of which the PIN and discriminator have wrong values.

2) Attempting to pair/control using Python controller fails to establish BLE connection and hence the EFR32 device is not recognized as a CHIP accessory.Here is the log -
```test@test:~$ sudo chip-device-ctrl 
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 112233
CHIP:DL: CHIP task running
Chip Device Controller Shell
chip-device-ctrl > ble-scan
2021-05-05 11:54:10,380 ChipBLEMgr   INFO     use default adapter
2021-05-05 11:54:10,515 ChipBLEMgr   INFO     scanning started
2021-05-05 11:54:20,630 ChipBLEMgr   INFO     scanning stopped
chip-device-ctrl > set-pairing-thread-credential 16 0x00da b3335216512091581d4ddc4b88eb9f53
Thread credential set
chip-device-ctrl > connect -ble 3840 73141520 4536
Device is assigned with nodeid = 4536
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:DL: TRACE: Bus acquired for name C-22c6
CHIP:DL: PlatformBlueZInit init success
CHIP:BLE: BLE scanning through known devices.
CHIP:BLE: Device C8:28:32:E2:15:76 does not look like a CHIP device.
CHIP:BLE: Device 63:98:C7:1F:9B:2A does not look like a CHIP device.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: Device 63:98:C7:1F:9B:2A does not look like a CHIP device.
CHIP:BLE: Device 44:0D:BF:1A:4E:72 does not look like a CHIP device.
CHIP:BLE: Device C8:28:32:E2:15:76 does not look like a CHIP device.
CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
Failed to establish secure session to device: 4050
Device temporary node id (**this does not match spec**): 4536
chip-device-ctrl > ```
 #### Proposed Solution
1) QR code string on the latest master should be fixed to generate expected PIN and discriminator value.
2) Python controller should be able to connect/pair EFR32 lock end device.

jmartinez-silabs, 2021-05-05:
Hi Saurab @saurabhst, I am starting to look into this. QRcode does have an isue on current master, but I am successful with the chip-device-ctrl. Did you update your commit for the controller and rebuilt it becasue it seems old.

`set-pairing-thread-credential` is not available anymore and should have returned a error. The version of your controller might be the issue for the `ble-connect`

```
chip-device-ctrl > set-pairing-thread-credential 16 0x00da b3335216512091581d4ddc4b88eb9f53
Pairing Thread Credential is nolonger available, use NetworkCommissioning cluster instead.
```

jmartinez-silabs, 2021-05-05:
Hey Saurab so I tried the master code. I think your issue is just compatibility issues using older commits for the Android Chip-tool and python controller.

I use current master (commit 674d1822662b71ee455821cf53355ea8fb1c62af) rebuild the EFR32 Light app, the Android Chip-tool and Python controller.

[image]
[image]

```
chip-device-ctrl > setup-payload parse-qr CH:Y3.1345600C0NF0ET00
Version: 0
VendorID: 9050
ProductID: 21324
RequiresCustomFlow: 0
RendezvousInformation: 2 [BLE]
Discriminator: 3840
SetUpPINCode: 73141520
```
```
chip-device-ctrl > connect -ble 3840 73141520 4538
Device is assigned with nodeid = 4538
CHIP:BLE: BLE removing known devices.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: New device scanned: 67:48:F8:96:5E:C3
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Scan complete notification without an active scan.
CHIP:BLE: subscribe complete, ep = 0xffff951c70e0
CHIP:BLE: peripheral chose BTP version 3; central expected between 2 and 3
CHIP:BLE: using BTP fragment sizes rx 20 / tx 128.
CHIP:BLE: local and remote recv window size = 3
Secure Session to Device Established
Device temporary node id (**this does not match spec**): 4538
chip-device-ctrl > 
```

Please rebuild the tools you use and let me know if this is still an issue for you

closed 2021-05-05 by jmartinez-silabs