project-chip/connectedhomeip #6506: Barcode from TE2 image produce error in iOS chip-tool

fdelacourt-wyze, 2021-05-05:
#### Problem
When launching the lighting-app example and generating a barcode or pasting the barcode in the chip-tool on iOS, an error is generated instead.
When the code is pasted and the hit button is pushed, I get an Integrity check failed. when I generate the barcode and then scan it, I get an Unexpected Integer Value.

<what's wrong or missing, please include any applicable:
 * expected behavior: barcode is ingested properly and the device information are displayed.
 * actual behavior: error is generated blocking the next step
 * steps to reproduce: 
 1 - Launch the lighting app example.
 Output for me:  
>  ./chip-lighting-app
[...]
SetupPINCode: [20202021]
SetupQRCode:  [CH:H34DV+700 V5C.71]
CHIP:ZCL: Using ZAP configuration...
2 - Copy the  QRCode
-> CH:H34DV+700 V5C.71

3 - Launch the CHIPTool app on iPhone
4 - Select QR code Parser and paste the code
5 - Hit the Go Button
Result: Received a Integrity Check Failed message

6 - Go to barcode generating site.
Used: https://www.the-qrcode-generator.com

7 - Paste the barcode copied from the lighting app as free text  and generate the barcode

8 - Go the CHIPTool app on iPhone and in select the camera in the QR code Parser

9 - Scan barcode generated
Result: Received a Unexpected Integer Value

woody-apple, 2022-01-25:
We're in progress of reviewing older issues. Is this still an issue given how old the SDK this was tested against? Please re-open if so. Thanks!

closed 2022-01-25 by woody-apple

fdelacourt-wyze, 2022-01-25:
No I think we can safely close that one. Sorry I should have cleaned it up when subsequent build fixed the issue.

Thank you,

Frederik Delacourt
—————————————
***@***.******@***.***>
763 639 4075

[A sign in the dark  Description automatically generated]