project-chip/connectedhomeip #6577: the nrfconnect lighting app is broken

wgtdkp, 2021-05-07:
#### Problem
commit: c9d22b0bb8bcfc9b7ef8abee1e9093765cb0fec7

The nrfconnect lighting app is in crash-rebooting loop:
```txt
E: ***** USAGE FAULT *****
E:   Unaligned memory access
E: r0/a1:  0x200217c0  r1/a2:  0x00046d49  r2/a3:  0x00046d39
E: r3/a4:  0x00046d49 r12/ip:  0x00000000 r14/lr:  0x00065361
E:  xpsr:  0x81000000
E: Faulting instruction address (r15/pc): 0x00065656
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20005ef0 (CHIP)
I: nRF5 802154 radio initialized
I: 8 Sectors of 4096 bytes                                                                                                                                   
I: alloc wra: 0, ff0                                                                                                                                         
I: data wra: 0, 0                                                                                                                                            
*** Booting Zephyr OS build v2.4.99-ncs1-1829-ga9d3aef65424  ***                                                                                             
I: Init CHIP stack                                                                                                                                           
I: 120 [DL]BLE address was set to D9:78:92:EA:C3:CA
I: SoftDevice Controller build revision: 
I: cf 11 3d 18 6d 33 fb cb |..=.m3..
I: d8 23 00 ab 74 a5 51 6f |.#..t.Qo
I: c0 69 f5 b2             |.i..    
I: Starting CHIP task
I: Init Thread stack
I: 147 [DL]OpenThread started: OK
I: 150 [DL]Setting OpenThread device type to MINIMAL END DEVICE
I: 156 [ZCL]Using ZAP configuration...
I: 160 [ZCL]deactivate report event
I: 163 [ZCL]0x78e43ep 1 clus  6 attr  0 not supported
E: Updating on/off cluster failed: 86
I: 171 [ZCL]0x78e43ep 1 clus  8 attr  0 not supported
E: Updating level cluster failed: 86
I: 179 [ZCL]Cluster callback: 40
I: 182 [ZCL]Unknown cluster ID: 40
I: 186 [ZCL]Cluster callback: 40
I: 189 [ZCL]Unknown cluster ID: 40
I: 192 [ZCL]Cluster callback: 40
I: 195 [ZCL]Unknown cluster ID: 40
I: 198 [ZCL]Cluster callback: 40
I: 201 [ZCL]Unknown cluster ID: 40
I: 204 [ZCL]Cluster callback: 40
I: 207 [ZCL]Unknown cluster ID: 40
I: 210 [ZCL]Cluster callback: 40
I: 213 [ZCL]Unknown cluster ID: 40
I: 217 [ZCL]Cluster callback: 40
I: 219 [ZCL]Unknown cluster ID: 40
I: 223 [ZCL]Cluster callback: 40
I: 226 [ZCL]Unknown cluster ID: 40
I: 229 [ZCL]Cluster callback: 40
I: 232 [ZCL]Unknown cluster ID: 40
I: 235 [DIS]Init admin pairing table with server storage.
D: 240 [IN]TransportMgr initialized
I: 244 [IN]local node id is 0x0000000000bc5c01
I: 248 [DIS]Start dns-sd server
I: 251 [SVR]Server Listening...
I: 254 [DL]Device Configuration:
I: 257 [DL]  Serial Number: TEST_SN
I: 260 [DL]  Vendor Id: 9050 (0x235A)
I: 264 [DL]  Product Id: 20044 (0x4E4C)
I: 267 [DL]  Product Revision: 1
I: 270 [DL]  Setup Pin Code: 20202021
I: 274 [DL]  Setup Discriminator: 3840 (0xF00)
I: 278 [DL]  Manufacturing Date: (not set)
I: 282 [SVR]SetupQRCode: [CH:Y3.1324600A50324800]
I: 286 [SVR]Copy/paste the below URL in a browser to see the QR Code:
I: 293 [SVR]https://dhrishi.github.io/connectedhomeip/qrcode.html?data=CH%3AY3.1324600A50324800
I: 301 [SVR]Manual pairing code: [34970112332]
D: 305 [DL]CHIP task running
E: ***** USAGE FAULT *****
E:   Unaligned memory access
E: r0/a1:  0x200217c0  r1/a2:  0x00046d49  r2/a3:  0x00046d39
E: r3/a4:  0x00046d49 r12/ip:  0x00000000 r14/lr:  0x00065361
E:  xpsr:  0x81000000
E: Faulting instruction address (r15/pc): 0x00065656
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20005ef0 (CHIP)
E: Resetting system
```

Damian-Nordic, 2021-05-07:
@wgtdkp I can't find `c9d22b0bb8bcfc9b7ef8abee1e9093765cb0fec7` commit in my tree, but I built and flashed the current master and I'm not seeing such an issue. Can you verify if TOT works for you? Does the commit belong to a private fork or you pasted a wrong SHA?

wgtdkp, 2021-05-08:
@Damian-Nordic This issue disappears after updating to commit 4718196aca9fdf1aa491195b807ddaade4ebf5b8. Thanks!

closed 2021-05-08 by wgtdkp