project-chip/connectedhomeip #6578: Pairing via Bypass (Rendezvous) Fails

kianooshkarami, 2021-05-07:
#### Problem
 * expected behavior: Ability to pair via Bypass method fails using (Tested with commit: ab8122e4e05f9757a9156c4f23f4a27f715a5bf2, Last commit that I tested this functionality and was successful: 92e3683cd0a2772b91c08e0b0109395f94944f1b), I can try to pinpoint the commit that may have introduced this regression a bit later.
 * actual behavior: Failure to pair with the device
 * steps to reproduce: `chip-tool pairing bypass {ip_addr} 11097`
 * system configuration: Linux (Ubuntu)
 * Logs:
 
```
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x000000000001b669
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:IN: New pairing for device 0x0000000000bc5c01, key 65535!!
CHIP:BLE: Failed in setting up secure channel: err CHIP Error 4003 (0x00000FA3): Incorrect state
CHIP:TOO: Secure Pairing Failed
CHIP:DL: CHIP task running
CHIP:TOO: Pairing Failure: CHIP Error 4003 (0x00000FA3): Incorrect state
CHIP:CTL: Shutting down the commissioner
CHIP:CTL: Shutting down the controller
CHIP:DL: System Layer shutdown
CHIP:DL: Inet Layer shutdown
CHIP:DL: BLE layer shutdown
```

 #### Proposed Solution

kianooshkarami, 2021-05-07:
Also, FWIW, I do not think `BLE` is the appropriate module for the following log statement. `CHIP:BLE: Failed in setting up secure channel: err CHIP Error 4003 (0x00000FA3): Incorrect state`

kianooshkarami, 2021-05-07:
I was able to pinpoint that this commit (0f443b8502f530ddb79e480d372ce5ec53749e01) broke the bypass functionality.

wissn84, 2021-05-10:
To test "bypass", You shall include flag both side

gn gen out/debug --args='chip_bypass_rendezvous=false'

closed 2021-05-12 by andy31415

nageshshamnur, 2021-05-18:
@kianooshkarami : Can you tell me how to get the remote port number. I am unable to find in logs while running all-clusters-app. Thanks

PR #6681 (2021-05-12): Fix rendezvous bypass mode
#### Problem
Rendezvous bypass mode is not working. It's failing in establishing the secure session.

 #### Summary of Changes
Recent controller changes mixed using test secret with non-test pairing. This change makes sure that the test secret is used for establishing the secure session when bypass mode is being used.

Fixes #6578