project-chip/connectedhomeip #6653: How do i use zap files

ChengqiangShe, 2021-05-11:
#### How do i use zap files
In the connectedhomeip/examples/lighting-app/lighting-common directory, you can see lighting-app.zap, but how should I use or modify this file.

markus-becker-tridonic-com, 2021-05-19:
You can use the script https://github.com/project-chip/connectedhomeip/blob/master/scripts/tools/zap/configure.sh to start the zap tool. You need to install a few dependencies beforehand: https://github.com/project-chip/zap/tree/f2ef0141a099cfb6226f88edb9d341fe594884ee#quick-instructions.

Once you opened the zap file, you can modify it to meet your cluster needs. Save it and run 
https://github.com/project-chip/connectedhomeip/blob/master/scripts/tools/zap_regen_all.py
which will re-generate the files in the gen/ folder.

markus-becker-tridonic-com, 2021-05-21:
@ChengqiangShe any other questions or would you close the issue?

closed 2021-05-21 by ChengqiangShe