project-chip/connectedhomeip #6661: Python controller fails  to complete BLE Connect to the device

Minan-art, 2021-05-11:
#### Problem
 >
To get pairing flow working on thread devices, use the following commands from the controller ：
connect -ble 3840 73141520 4536

As shown below, able to  see  the Secure Session to Device Established,  but Python controller continues to send messages of HandlePlatformSpecificBLEEvent 16387,  hence I couldn't execute any commands to complete pairing.
[image]

Any explanation and solution?

Here is the log -

chip-device-ctrl > ### **ble-scan**
2021-05-10 07:45:06,280 ChipBLEMgr   INFO     scanning started
2021-05-10 07:45:06,433 ChipBLEMgr   INFO     Name            = EFR32_LIGHT
2021-05-10 07:45:06,434 ChipBLEMgr   INFO     ID              = d558231f-fb05-389c-bae2-6b797b364d28
2021-05-10 07:45:06,435 ChipBLEMgr   INFO     RSSI            = -25
2021-05-10 07:45:06,437 ChipBLEMgr   INFO     Address         = 60:7F:63:3E:17:94
2021-05-10 07:45:06,454 ChipBLEMgr   INFO     Pairing State   = 0
2021-05-10 07:45:06,455 ChipBLEMgr   INFO     Discriminator   = 3840
2021-05-10 07:45:06,455 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-05-10 07:45:06,455 ChipBLEMgr   INFO     Product Id      = 21324
2021-05-10 07:45:06,459 ChipBLEMgr   INFO     Adv UUID        = 0000fff6-0000-1000-8000-00805f9b34fb
2021-05-10 07:45:06,460 ChipBLEMgr   INFO     Adv Data        = 00000f5a234c53
2021-05-10 07:45:06,460 ChipBLEMgr   INFO
2021-05-10 07:45:16,983 ChipBLEMgr   INFO     scanning stopped
chip-device-ctrl > **connect -ble 3840 73141520 4536**
Device is assigned with nodeid = 4536
CHIP:BLE: Assigned local session key ID 1
CHIP:EM: SessionEstablishmentExchangeDispatch::SendMessageImpl  mTransportMgr 0xd713d40
CHIP:IN: Message appended to BLE send queue
CHIP:BLE: Sent PBKDF param request
CHIP:BLE: BLE removing known devices.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: Device DC:23:4D:0D:69:87 does not look like a CHIP device.
CHIP:BLE: Device DC:23:4D:0D:6F:47 does not look like a CHIP device.
CHIP:BLE: Device E8:97:CA:76:89:70 does not look like a CHIP device.
CHIP:BLE: Device DC:23:4D:0D:6E:0A does not look like a CHIP device.
CHIP:BLE: Device DC:23:4D:0D:6D:6B does not look like a CHIP device.
CHIP:BLE: Device 53:DF:FB:AA:90:6A does not look like a CHIP device.
CHIP:BLE: Device DC:23:4D:0D:6C:95 does not look like a CHIP device.
CHIP:BLE: Device 84:E3:42:AE:79:48 does not look like a CHIP device.
CHIP:BLE: Device 42:3A:28:12:8D:14 does not look like a CHIP device.
CHIP:BLE: Device DC:23:4D:0D:6A:B4 does not look like a CHIP device.
CHIP:BLE: Device 5B:01:18:D1:43:70 does not look like a CHIP device.
CHIP:BLE: Device 4B:AA:F7:2E:DF:08 does not look like a CHIP device.
CHIP:BLE: Device DC:23:4D:0D:6A:79 does not look like a CHIP device.
CHIP:BLE: Device DC:23:4D:0E:03:B2 does not look like a CHIP device.
CHIP:BLE: Device DC:23:4D:21:E4:8C does not look like a CHIP device.
CHIP:BLE: Device 68:36:B4:7E:BA:5F does not look like a CHIP device.
CHIP:BLE: Device DC:23:4D:0E:04:6B does not look like a CHIP device.
CHIP:BLE: Device 40:E0:23:7B:A0:F5 does not look like a CHIP device.
CHIP:BLE: Device DC:23:4D:0D:6A:EA does not look like a CHIP device.
CHIP:BLE: Device 75:D8:50:5E:BC:57 does not look like a CHIP device.
CHIP:BLE: Device DC:23:4D:0D:6D:83 does not look like a CHIP device.
CHIP:BLE: New device scanned: 75:B8:14:05:B9:51
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Scan complete notification without an active scan.
CHIP:DL: ConnectDevice complete
CHIP:DL: Char1 /org/bluez/hci0/dev_75_B8_14_05_B9_51/service0001
CHIP:DL: Char1 /org/bluez/hci0/dev_75_B8_14_05_B9_51/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_75_B8_14_05_B9_51/service0001
CHIP:DL: Char1 /org/bluez/hci0/dev_75_B8_14_05_B9_51/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_75_B8_14_05_B9_51/service0001
CHIP:DL: Char1 /org/bluez/hci0/dev_75_B8_14_05_B9_51/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_75_B8_14_05_B9_51/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_75_B8_14_05_B9_51/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_75_B8_14_05_B9_51/service0001
CHIP:DL: Char1 /org/bluez/hci0/dev_75_B8_14_05_B9_51/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_75_B8_14_05_B9_51/service0001
[... 71 lines cut ...]
CHIP:BLE: Sent spake2p msg3
CHIP:IN: New pairing for device 0x00000000000011b8, key 0!!
CHIP:CTL: Remote device completed SPAKE2+ handshake

CHIP:CTL: AsyncSetKeyValue: PairedDevice11b8=EAC8Wgh2Yj9b7rBHty/Nn0X5AAAAAAAAAAAAAAAAAAAAAAEAAQAAALgRAAAAAAAANDBjNDpiZGUwOmZmZmY6MDpiMGY1OjUyYWE6ZmZmZjowAAAAAAAAAAAAAAAAAFkrAAACAAAAAAAAAAAAAAAAAAAAAAAAAAAA
CHIP:CTL: AsyncSetKeyValue: ListPairedDevices0=uBEAAAAAAAA=
CHIP:CTL: AsyncSetKeyValue: StartKeyID=2
Secure Session to Device Established
Device temporary node id (**this does not match spec**): 4536
chip-device-ctrl > **CHIP:DL: HandlePlatformSpecificBLEEvent 16387**
CHIP:DL: Indication received, conn = 0xffffa004cea0
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffffa004cea0
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffffa004cea0
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffffa004cea0
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffffa004cea0
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387

Minan-tuya, 2021-05-11:
@jmartinez-silabs 
If  the BLE keeps continue having some TX RX with the devices，Python controller will continue to send HandlePlatformSpecificBLEEvent 16387.  Hence I couldn‘t  send the next zcl commands because  I couldn‘t  switch to the terminal of  chip-device-ctrl .

jmartinez-silabs, 2021-05-11:
@Minan-tuya That is incorrect. You close the ble session with close-ble after the zcl NetworkCommissioning AddThreadNetwork  and zcl NetworkCommissioning EnableNetwork command have been successfully sent

Here is a log of my controller
```
sudo out/python_env/bin/chip-device-ctrl
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x000000000001b669
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:DL: CHIP task running
CHIP:DL: Platform main loop started.
Chip Device Controller Shell

chip-device-ctrl > connect -ble 3840 73141520 4536
Device is assigned with nodeid = 4536
CHIP:BLE: Assigned local session key ID 0
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:DL: CHIP:IN: TRACE: Bus acquired for name C-af3eMessage appended to BLE send queue
CHIP:BLE: Sent PBKDF param request

CHIP:CTL: SyncSetKeyValue on StartKeyID
CHIP:DL: PlatformBlueZInit init success
CHIP:BLE: BLE removing known devices.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: New device scanned: 55:CD:9D:13:39:61
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Scan complete notification without an active scan.
CHIP:DL: ConnectDevice complete
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service000e
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service000e
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service000e
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0001
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0001
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0001
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service000e
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service000e
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service000e
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0015
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0015
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0015
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_55_CD_9D_13_39_61/service0018
CHIP:DL: New BLE connection 0xffffb003ebe0, device 55:CD:9D:13:39:61, path /org/bluez/hci0/dev_55_CD_9D_13_39_61
CHIP:DL: HandlePlatformSpecificBLEEvent 16385
CHIP:IN: BleConnectionComplete: endPoint 0xffffbc91cda8
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: HandlePlatformSpecificBLEEvent 16388
CHIP:BLE: subscribe complete, ep = 0xffffbc91cda8
CHIP:DL: Indication received, conn = 0xffffb003ebe0
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:BLE: peripheral chose BTP version 3; central expected between 2 and 3
CHIP:BLE: using BTP fragment sizes rx 20 / tx 128.
CHIP:BLE: local and remote recv window size = 3
CHIP:IN: BLE EndPoint 0xffffbc91cda8 Connection Complete
[... 124 lines cut ...]
CHIP:ZCL: DefaultResponse:
CHIP:ZCL:   ClusterId: 0x0031
CHIP:ZCL:   CommandId: 0x0c
CHIP:ZCL:   status: EMBER_ZCL_STATUS_SUCCESS (0x00)
Failed to establish secure session to device: 3198972256
CHIP:ZCL: Data model processing success!
An exception occurred during process ZCL command:
Error -1095995040 (0xBEAC7160)
chip-device-ctrl > CHIP:DL: Indication received, conn = 0xffffb003ebe0
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffffb003ebe0
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
close-ble
CHIP:IN: Clearing BLE pending packets.
CHIP:BLE: Auto-closing end point's BLE connection.
CHIP:DL: Closing BLE GATT connection (con 0xffffb003ebe0)
CHIP:DL: BluezDisconnect peer=55:CD:9D:13:39:61
chip-device-ctrl > CHIP:DL: HandlePlatformSpecificBLEEvent 16388
CHIP:BLE: no endpoint for unsub complete
CHIP:DL: Bluez disconnected
CHIP:DL: Bluez notify CHIPoBluez connection disconnected

```

closed 2021-06-17 by jmartinez-silabs