project-chip/connectedhomeip #6711: TestCaseSession.cpp fails with is_asan=true

vivien-apple, 2021-05-12:
```
READ of size 20 at 0x616000000af2 thread T0
    #0 0x102d695b5 in MemcmpInterceptorCommon(void*, int (*)(void const*, void const*, unsigned long), void const*, void const*, unsigned long)+0x155 (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x195b5)
    #1 0x102d69ab9 in wrap_memcmp+0x69 (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x19ab9)
    #2 0x10294985d in chip::Credentials::CertificateKeyId::IsEqual(chip::Credentials::CertificateKeyId const&) const CHIPCert.cpp:787
    #3 0x10294ae7f in chip::Credentials::ChipCertificateSet::FindValidCert(chip::Credentials::ChipDN const&, chip::Credentials::CertificateKeyId const&, chip::Credentials::ValidationContext&, chip::BitFlags<chip::Credentials::CertValidateFlags, unsigned char>, unsigned char, chip::Credentials::ChipCertificateData*&) CHIPCert.cpp:570
    #4 0x10294aa6f in chip::Credentials::ChipCertificateSet::FindValidCert(chip::Credentials::ChipDN const&, chip::Credentials::CertificateKeyId const&, chip::Credentials::ValidationContext&, chip::Credentials::ChipCertificateData*&) CHIPCert.cpp:372
    #5 0x102956561 in chip::Credentials::OperationalCredentialSet::FindValidCert(chip::Credentials::CertificateKeyId const&, chip::Credentials::ChipDN const&, chip::Credentials::CertificateKeyId const&, chip::Credentials::ValidationContext&, chip::Credentials::ChipCertificateData*&) CHIPOperationalCredentials.cpp:206
    #6 0x1029ae409 in chip::CASESession::Validate_and_RetrieveResponderID(unsigned char const**, chip::Crypto::P256PublicKey&, unsigned char const**, unsigned short&) CASESession.cpp:1015
    #7 0x1029ac053 in chip::CASESession::HandleSigmaR2(chip::System::PacketBufferHandle const&) CASESession.cpp:653
    #8 0x1029ab524 in chip::CASESession::HandleSigmaR2_and_SendSigmaR3(chip::System::PacketBufferHandle const&) CASESession.cpp:557
    #9 0x1029b0895 in chip::CASESession::OnMessageReceived(chip::Messaging::ExchangeContext*, chip::PacketHeader const&, chip::PayloadHeader const&, chip::System::PacketBufferHandle) CASESession.cpp:1152
    #10 0x10297cbd4 in chip::Messaging::ExchangeContext::HandleMessage(chip::PacketHeader const&, chip::PayloadHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) ExchangeContext.cpp:419
    #11 0x102981c60 in chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SecureSessionHandle, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle, chip::SecureSessionMgr*) ExchangeMgr.cpp:243
    #12 0x10299da21 in chip::SecureSessionMgr::MessageDispatch(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) SecureSessionMgr.cpp:317
    #13 0x10299c4ec in chip::SecureSessionMgr::OnMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) SecureSessionMgr.cpp:306
    #14 0x1029a047a in chip::TransportMgrBase::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) TransportMgrBase.cpp:57
    #15 0x1029447ed in chip::Transport::Base::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) Base.h:92
    #16 0x1029445bd in LoopbackTransport::SendMessage(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) TestCASESession.cpp:59
    #17 0x10299ff9d in chip::TransportMgrBase::SendMessage(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) TransportMgrBase.cpp:27
    #18 0x1029b23db in chip::SessionEstablishmentExchangeDispatch::SendMessageImpl(chip::SecureSessionHandle, chip::PayloadHeader&, chip::System::PacketBufferHandle&&, chip::EncryptedPacketBufferHandle*) SessionEstablishmentExchangeDispatch.cpp:42
    #19 0x10297df04 in chip::Messaging::ExchangeMessageDispatch::SendMessage(chip::SecureSessionHandle, unsigned short, bool, chip::Messaging::ReliableMessageContext*, bool, chip::Protocols::Id, unsigned char, chip::System::PacketBufferHandle) ExchangeMessageDispatch.cpp:95
    #20 0x10297b063 in chip::Messaging::ExchangeContext::SendMessageImpl(chip::Protocols::Id, unsigned char, chip::System::PacketBufferHandle, chip::BitFlags<chip::Messaging::SendMessageFlags, unsigned short> const&, chip::Transport::PeerConnectionState*) ExchangeContext.cpp:163
    #21 0x10297a75f in chip::Messaging::ExchangeContext::SendMessage(chip::Protocols::Id, unsigned char, chip::System::PacketBufferHandle, chip::BitFlags<chip::Messaging::SendMessageFlags, unsigned short> const&) ExchangeContext.cpp:107
    #22 0x1029a83e8 in int chip::Messaging::ExchangeContext::SendMessage<chip::Protocols::SecureChannel::MsgType, void>(chip::Protocols::SecureChannel::MsgType, chip::System::PacketBufferHandle&&, chip::BitFlags<chip::Messaging::SendMessageFlags, unsigned short> const&) ExchangeContext.h:105
    #23 0x1029aa264 in chip::CASESession::SendSigmaR2() CASESession.cpp:540
    #24 0x1029a8522 in chip::CASESession::HandleSigmaR1_and_SendSigmaR2(chip::System::PacketBufferHandle const&) CASESession.cpp:348
    #25 0x1029b0880 in chip::CASESession::OnMessageReceived(chip::Messaging::ExchangeContext*, chip::PacketHeader const&, chip::PayloadHeader const&, chip::System::PacketBufferHandle) CASESession.cpp:1148
    #26 0x10297cbd4 in chip::Messaging::ExchangeContext::HandleMessage(chip::PacketHeader const&, chip::PayloadHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) ExchangeContext.cpp:419
    #27 0x10298237f in chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SecureSessionHandle, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle, chip::SecureSessionMgr*) ExchangeMgr.cpp:308
    #28 0x10299da21 in chip::SecureSessionMgr::MessageDispatch(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) SecureSessionMgr.cpp:317
    #29 0x10299c4ec in chip::SecureSessionMgr::OnMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) SecureSessionMgr.cpp:306
    #30 0x1029a047a in chip::TransportMgrBase::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) TransportMgrBase.cpp:57
    #31 0x1029447ed in chip::Transport::Base::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) Base.h:92
    #32 0x1029445bd in LoopbackTransport::SendMessage(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) TestCASESession.cpp:59
    #33 0x10299ff9d in chip::TransportMgrBase::SendMessage(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) TransportMgrBase.cpp:27
    #34 0x1029b23db in chip::SessionEstablishmentExchangeDispatch::SendMessageImpl(chip::SecureSessionHandle, chip::PayloadHeader&, chip::System::PacketBufferHandle&&, chip::EncryptedPacketBufferHandle*) SessionEstablishmentExchangeDispatch.cpp:42
    #35 0x10297df04 in chip::Messaging::ExchangeMessageDispatch::SendMessage(chip::SecureSessionHandle, unsigned short, bool, chip::Messaging::ReliableMessageContext*, bool, chip::Protocols::Id, unsigned char, chip::System::PacketBufferHandle) ExchangeMessageDispatch.cpp:95
    #36 0x10297b063 in chip::Messaging::ExchangeContext::SendMessageImpl(chip::Protocols::Id, unsigned char, chip::System::PacketBufferHandle, chip::BitFlags<chip::Messaging::SendMessageFlags, unsigned short> const&, chip::Transport::PeerConnectionState*) ExchangeContext.cpp:163
    #37 0x10297a75f in chip::Messaging::ExchangeContext::SendMessage(chip::Protocols::Id, unsigned char, chip::System::PacketBufferHandle, chip::BitFlags<chip::Messaging::SendMessageFlags, unsigned short> const&) ExchangeContext.cpp:107
    #38 0x1029a83e8 in int chip::Messaging::ExchangeContext::SendMessage<chip::Protocols::SecureChannel::MsgType, void>(chip::Protocols::SecureChannel::MsgType, chip::System::PacketBufferHandle&&, chip::BitFlags<chip::Messaging::SendMessageFlags, unsigned short> const&) ExchangeContext.h:105
    #39 0x1029a7255 in chip::CASESession::SendSigmaR1() CASESession.cpp:337
    #40 0x1029a686a in chip::CASESession::EstablishSession(chip::Transport::PeerAddress, chip::Credentials::OperationalCredentialSet*, unsigned long long, unsigned short, chip::Messaging::ExchangeContext*, chip::SessionEstablishmentDelegate*) CASESession.cpp:240
[... cut ...]
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==29092==ABORTING
'#0:','Test-CHIP-SecurePairing'
```

bzbarsky-apple, 2022-06-13:
@vivien-apple I just tried:
```
./gn_build.sh enable_host_clang_build=false enable_host_gcc_mbedtls_build=false is_asan=true
```
and I am not seeing this failure.  Is this still happening?

andy31415, 2022-09-28:
No updates since June, we run tests using ASAN and TSAN on linux and darwin and have not seen this kind of failure. I assume it was fixed since 1.5 years ago.

closed 2022-09-28 by andy31415