project-chip/connectedhomeip #6726: OnSrpClientNotification: Operation refused for security reasons

jmartinez-silabs, 2021-05-12:
#### Problem
When the Global Ipv6 address is obtained after becoming Child on the thread network. The srp services fails the update as the server on BR refuse the action.

The RemoveSrpService is succesfull as we see in the BR trace that th service is unpublished. But following the AddSrpService fails on the BR side with an error code 
```
May 12 10:59:04 raspberrypi otbr-agent[429]: [INFO]-SRP-----: [server] outstanding service update timeout
May 12 10:59:04 raspberrypi otbr-agent[429]: [INFO]-SRP-----: [server] send fail response: 5
```
 
DeviceLog
```
<detail> [DL]    Device Role: DETACHED
<detail> [DL]    Network Name: OpenThreadDemo
<detail> [DL]    PAN Id: 0x1234
<detail> [DL]    Extended PAN Id: 0x1111111122222222
<detail> [DL]    Channel: 15
<detail> [DL]    Mesh Prefix: FDBD:BE31:E3EA:E982::/64
<detail> [DL]    Thread Unicast Addresses:
<detail> [DL]         FDBD:BE31:E3EA:E982:CF70:FA5A:9D6:309A/64 valid preferred
<detail> [DL]         FE80::E4B4:C901:6EC1:4051/64 valid preferred
<detail> [DL] LwIP Thread interface UP
<detail> [DL] LwIP Thread interface addresses updated
<detail> [DL]    FE80::E4B4:C901:6EC1:4051 IPv6 link-local address
<info  > [DIS] Found admin paring for admin 0, node 0x00000000000011b8
<detail> [DIS] Using Thread extended MAC for hostname.
<info  > [DIS] Advertise operational node 0x0000000000000000-0x00000000000011b8
...
<info  > [DL] SRP Client was started, as detected server addressed: fdbd:be31:e3ea:e982:0:ff:fe00:4400
<info  > [DL] _OnPlatformEvent default:  event->Type = 32778
<detail> [DL] OpenThread State Changed (Flags: 0x200012a4)
<detail> [DL]    Device Role: CHILD
<detail> [DL]    Partition Id: 0x70A15C94
<info  > [DL] _OnPlatformEvent default:  event->Type = 32778
<detail> [DL] OpenThread State Changed (Flags: 0x00000001)
<detail> [DL]    Thread Unicast Addresses:
<detail> [DL]         FD11:22::DA01:9A4D:5D07:D6FA/64 valid preferred
<detail> [DL]         FDBD:BE31:E3EA:E982:0:FF:FE00:4401/64 valid preferred rloc
<detail> [DL]         FDBD:BE31:E3EA:E982:CF70:FA5A:9D6:309A/64 valid preferred
<detail> [DL]         FE80::E4B4:C901:6EC1:4051/64 valid preferred
<detail> [DL] LwIP Thread interface addresses updated
<detail> [DL]    FE80::E4B4:C901:6EC1:4051 IPv6 link-local address, preferred)
<detail> [DL]    FDBD:BE31:E3EA:E982:CF70:FA5A:9D6:309A Thread mesh-local address)
<detail> [DL]    FD11:22::DA01:9A4D:5D07:D6FA IPv6 unique local address, preferred)
<info  > [DIS] Found admin paring for admin 0, node 0x00000000000011b8
<detail> [DIS] Using Thread extended MAC for hostname.
<info  > [DIS] Advertise operational node 0x0000000000000000-0x00000000000011b8
<error > [SVR] Failed to start operational advertising: OpenThread Error 9000029 (0x0089545D): Duplicated
<error > [DL] Long dispatch time: 123 ms
<error > [DL] OnSrpClientNotification: Operation refused for security reasons
<error > [DL] OnSrpClientNotification: Operation refused for security reasons
<error > [DL] OnSrpClientNotification: Operation refused for security reasons
<error > [DL] OnSrpClientNotification: Operation refused for security reasons
```

OTBR log
```
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-SRP-----: [server] receive DNS update from fdbd:be31:e3ea:e982:0:ff:fe00:4401
May 12 10:59:03 raspberrypi otbr-agent[429]: [adproxy] advertise SRP service updates: host=E6B4C9016EC14051.default.service.arpa.
May 12 10:59:03 raspberrypi otbr-agent[429]: [adproxy] publish SRP host: E6B4C9016EC14051.default.service.arpa.
May 12 10:59:03 raspberrypi otbr-agent[429]: [mdns] publish new host E6B4C9016EC14051
May 12 10:59:03 raspberrypi otbr-agent[429]: [mdns] add new host E6B4C9016EC14051
May 12 10:59:03 raspberrypi otbr-agent[429]: [adproxy] publish SRP service: 0000000000000000-00000000000011B8._chip._tcp.default.service.arpa.
May 12 10:59:03 raspberrypi otbr-agent[429]: [mdns] add service: 0000000000000000-00000000000011B8._chip._tcp (ref: 0x789680)
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-MAC-----: Frame rx failed, error:Duplicated, len:122, seqnum:249, type:Data, src:0x4401, dst:0x4400, sec:yes, ackreq:yes
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-MAC-----: Received IPv6 UDP msg, len:412, chksum:57eb, from:0x4401, sec:yes, prio:normal, rss:-76.0
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-MAC-----:     src:[fdbd:be31:e3ea:e982:0:ff:fe00:4401]:49152
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-MAC-----:     dst:[fdbd:be31:e3ea:e982:0:ff:fe00:4400]:49152
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-PLAT----: processReceive: OK
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-SRP-----: [server] receive DNS update from fdbd:be31:e3ea:e982:0:ff:fe00:4401
May 12 10:59:03 raspberrypi otbr-agent[429]: [adproxy] advertise SRP service updates: host=E6B4C9016EC14051.default.service.arpa.
May 12 10:59:03 raspberrypi otbr-agent[429]: [adproxy] publish SRP host: E6B4C9016EC14051.default.service.arpa.
May 12 10:59:03 raspberrypi otbr-agent[429]: [mdns] update existing host E6B4C9016EC14051
May 12 10:59:03 raspberrypi otbr-agent[429]: [mdns] update existing host E6B4C9016EC14051
May 12 10:59:03 raspberrypi otbr-agent[429]: [adproxy] handle publish host 'E6B4C9016EC14051' result: 0
May 12 10:59:03 raspberrypi otbr-agent[429]: [adproxy] unpublish SRP service: 0000000000000000-00000000000011B8._chip._tcp.default.service.arpa.
May 12 10:59:03 raspberrypi otbr-agent[429]: [mdns] remove service ref 0x789680
May 12 10:59:03 raspberrypi mDNSResponder: mDNSCoreReceiveResponse: Received from 192.168.2.52:5353  112 OpenThreadDemo._meshcop._udp.local. TXT rv=1¦nn=OpenThreadDemo¦xp=#021#021#021#021""""¦tv=1.2.0¦dd=Z\}???/?¦sb=
May 12 10:59:03 raspberrypi mDNSResponder: mDNSCoreReceiveResponse: Resetting to Probing:   25 OpenThreadDemo._meshcop._udp.local. SRV 0 0 49191 raspberrypi.local.
May 12 10:59:03 raspberrypi mDNSResponder: Excessive name conflicts (72) for OpenThreadDemo._meshcop._udp.local. (SRV); rate limiting in effect
May 12 10:59:03 raspberrypi otbr-agent[429]: [NOTE]-ARP-----: Cache entry added: fdbd:be31:e3ea:e982:cf70:fa5a:9d6:309a, 0x4401 - snoop
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-MAC-----: Received IPv6 HopOpts msg, len:91, chksum:0000, from:0x4401, sec:yes, prio:normal, rss:-76.0
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-MAC-----:     src:[fdbd:be31:e3ea:e982:cf70:fa5a:9d6:309a]
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-MAC-----:     dst:[ff03:0:0:0:0:0:0:1]
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-PLAT----: processReceive: OK
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-PLAT----: MulticastRoutingManager: AddMulticastForwardingCache: add dynamic route: Thread fdbd:be31:e3ea:e982:cf70:fa5a:9d6:309a => ff03:0:0:0:0:0:0:1 None: OK
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-PLAT----: MulticastRoutingManager: ProcessMulticastRouterMessages: OK
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-MAC-----: Sent IPv6 HopOpts msg, len:91, chksum:0000, to:0xffff, sec:yes, prio:normal
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-MAC-----:     src:[fdbd:be31:e3ea:e982:cf70:fa5a:9d6:309a]
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-MAC-----:     dst:[ff03:0:0:0:0:0:0:1]
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-MAC-----: Sent IPv6 HopOpts msg, len:91, chksum:0000, to:0xffff, sec:yes, prio:normal
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-MAC-----:     src:[fdbd:be31:e3ea:e982:cf70:fa5a:9d6:309a]
May 12 10:59:03 raspberrypi otbr-agent[429]: [INFO]-MAC-----:     dst:[ff03:0:0:0:0:0:0:1]
May 12 10:59:04 raspberrypi otbr-agent[429]: [INFO]-SRP-----: [server] outstanding service update timeout
May 12 10:59:04 raspberrypi otbr-agent[429]: [INFO]-SRP-----: [server] send fail response: 5
May 12 10:59:04 raspberrypi otbr-agent[429]: [INFO]-PLAT----: processTransmit: OK
May 12 10:59:04 raspberrypi otbr-agent[429]: [INFO]-BR------: infra interface state changed: RUNNING, link-local-addr=fe80:0:0:0:dea6:32ff:fe9b:6863
May 12 10:59:04 raspberrypi otbr-agent[429]: [INFO]-MAC-----: Sent IPv6 UDP msg, len:60, chksum:eaa6, to:0x4401, sec:yes, prio:normal
May 12 10:59:04 raspberrypi otbr-agent[429]: [INFO]-MAC-----:     src:[fdbd:be31:e3ea:e982:0:ff:fe00:4400]:49152
May 12 10:59:04 raspberrypi otbr-agent[429]: [INFO]-MAC-----:     dst:[fdbd:be31:e3ea:e982:0:ff:fe00:4401]:49152
May 12 10:59:04 raspberrypi otbr-agent[429]: [INFO]-SRP-----: [server] outstanding service update timeout
May 12 10:59:04 raspberrypi otbr-agent[429]: [INFO]-SRP-----: [server] send fail response: 5
May 12 10:59:04 raspberrypi otbr-agent[429]: [INFO]-PLAT----: processTransmit: OK
```

jmartinez-silabs, 2021-05-12:
I have this issue on the EFR32 Platform. I spoke with @doru91 and he witnesses the same behaviour on NXP platofrm.
@kkasperczyk-no Do you see this behaviour on your platform? Before the RemoveSrpServices we had the duplicate error but the Service was published and was able to resolve the IP wth the advertissed info.

saurabhst, 2021-05-12:
I am also facing similar issue on the EFR32 Platform . OTBR is running on the Ubuntu VM with `nRF52840` connected .
OTBR logs
```
May 11 11:43:43 test otbr-agent[5994]: [INFO]-MAC-----: Received IPv6 UDP msg, len:412, chksum:402d, from:0xe401, sec:yes, prio:normal, rss:-43.375
May 11 11:43:43 test otbr-agent[5994]: [INFO]-MAC-----:     src:[fd9b:40e:34c4:d324:0:ff:fe00:e401]:49152
May 11 11:43:43 test otbr-agent[5994]: [INFO]-MAC-----:     dst:[fd9b:40e:34c4:d324:0:ff:fe00:e400]:49153
May 11 11:43:43 test otbr-agent[5994]: [INFO]-PLAT----: processReceive: OK
May 11 11:43:43 test otbr-agent[5994]: [INFO]-SRP-----: [server] receive DNS update from fd9b:40e:34c4:d324:0:ff:fe00:e401
May 11 11:43:43 test otbr-agent[5994]: [INFO]-ADPROXY-: Advertise SRP service updates: host=16B32A49556501C3.default.service.arpa.
May 11 11:43:43 test otbr-agent[5994]: [INFO]-ADPROXY-: Publish SRP host: 16B32A49556501C3.default.service.arpa.
May 11 11:43:43 test otbr-agent[5994]: [INFO]-MDNS----: Update existing host 16B32A49556501C3
May 11 11:43:43 test otbr-agent[5994]: [INFO]-MDNS----: Update existing host 16B32A49556501C3
May 11 11:43:43 test otbr-agent[5994]: [INFO]-ADPROXY-: Handle publish host '16B32A49556501C3' result: 0
May 11 11:43:43 test otbr-agent[5994]: [INFO]-SRP-----: [server] delayed SRP host update result, the SRP update has been committed
May 11 11:43:43 test otbr-agent[5994]: [INFO]-ADPROXY-: Unpublish SRP service: 0000000000000000-00000000000011C2._chip._tcp.default.service.arpa.
May 11 11:43:44 test otbr-agent[5994]: [INFO]-SRP-----: [server] outstanding service update timeout
May 11 11:43:44 test otbr-agent[5994]: [INFO]-SRP-----: [server] send fail response: 5
May 11 11:43:44 test otbr-agent[5994]: [INFO]-PLAT----: processTransmit: OK
```

EFR32 end device side Logs
```
<info  > [DL]  stop soft timer
<info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
<error > [DL] OnSrpClientNotification: Operation refused for security reasons
<info  > [DL] CHIPoBLE unsubscribe received
<info  > [DL] _OnPlatformEvent kCHIPoBLEUnsubscribe
<detail> [IN] Clearing BLE pending packets.
<info  > [BLE] Releasing end point's BLE connection back to application.
<info  > [DL] Disconnect Event for handle : 1
<info  > [DL] BLE GATT connection closed (con 1, reason 4115)
<info  > [DL] _OnPlatformEvent kCHIPoBLEConnectionError
<error > [DL] OnSrpClientNotification: Operation refused for security reasons
<error > [DL] OnSrpClientNotification: Operation refused for security reasons
<error > [DL] OnSrpClientNotification: Operation refused for security reasons
<info  > [DL] _OnPlatformEvent default:  event->Type = 32778
<detail> [DL] OpenThread State Changed (Flags: 0x00000064)
<detail> [DL]    Device Role: ROUTER
<error > [DL] OnSrpClientNotification: Operation refused for security reasons
<error > [DL] OnSrpClientNotification: Operation refused for security reasons
<error > [DL] OnSrpClientNotification: Operation refused for security reasons
<error > [DL] OnSrpClientNotification: Operation refused for security reasons
<info  > [DL] SRP Client was stopped, because current server is no longer detected.
```

kkasperczyk-no, 2021-05-13:
I can see on my platform different problem, looks like first few operations are successful, but then communication dies:
I: 113096 [DL]SRP Client was started, as detected server addressed: fdc6:9221:ba6:6037:0:ff:fe00:fc10
D: 113106 [DL]OpenThread State Changed (Flags: 0x301132b7)
D: 113111 [DL]   Device Role: CHILD
D: 113114 [DL]   Network Name: OpenThreadDemo
D: 113118 [DL]   PAN Id: 0x1234
D: 113121 [DL]   Extended PAN Id: 0x1111111122222222
D: 113126 [DL]   Channel: 15
D: 113129 [DL]   Mesh Prefix: fdc6:9221:ba6:6037::/64
D: 113134 [DL]   Partition Id: 0x7952225
D: 113137 [DL]   Thread Unicast Addresses:
D: 113141 [DL]        fd11:22::d62d:d2d:b2c8:5808/64 valid preferred
D: 113147 [DL]        fdc6:9221:ba6:6037:0:ff:fe00:1801/64 valid preferred rloc
D: 113155 [DL]        fdc6:9221:ba6:6037:745e:6a:9d2d:c9f0/64 valid preferred
D: 113162 [DL]        fe80::b4f7:2c56:b064:d896/64 valid preferred
E: 113170 [DIS]Fabric ID not known. Using a default
I: 113175 [DIS]Found admin paring for admin 0, node 0x0000000000000001
D: 113181 [DIS]Using Thread extended MAC for hostname.
I: 113186 [DIS]Advertise operational node 000000014a77cbb3-0000000000000001
I: 113193 [DL]removing srp service: 000000014A77CBB3-0000000000000001._chip._tcp
I: 113200 [DL]OnSrpClientNotification: Last requested operation completed successfully
I: 113208 [DL]advertising srp service: 000000014A77CBB3-0000000000000001._chip._tcp
E: 113215 [DL]Long dispatch time: 109 ms
D: 113219 [DL]OpenThread State Changed (Flags: 0x00000001)
D: 113224 [DL]   Thread Unicast Addresses:
D: 113228 [DL]        fd11:22::d62d:d2d:b2c8:5808/64 valid preferred
D: 113234 [DL]        fdc6:9221:ba6:6037:0:ff:fe00:1801/64 valid preferred rloc
D: 113242 [DL]        fdc6:9221:ba6:6037:745e:6a:9d2d:c9f0/64 valid preferred
D: 113249 [DL]        fe80::b4f7:2c56:b064:d896/64 valid preferred
E: 113257 [DIS]Fabric ID not known. Using a default
I: 113261 [DIS]Found admin paring for admin 0, node 0x0000000000000001
D: 113268 [DIS]Using Thread extended MAC for hostname.
I: 113273 [DIS]Advertise operational node 000000014a77cbb3-0000000000000001
I: 113280 [DL]removing srp service: 000000014A77CBB3-0000000000000001._chip._tcp
I: 113287 [DL]OnSrpClientNotification: Last requested operation completed successfully
I: 113295 [DL]advertising srp service: 000000014A77CBB3-0000000000000001._chip._tcp
D: 114723 [DL]Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 3)
D: 114816 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
E: 115212 [DL]OnSrpClientNotification: Timed out waiting on server response
D: 117292 [DL]ConnId: 0x00, New CCCD value: 0x0000
D: 117297 [IN]Clearing BLE pending packets.
I: 117301 [BLE]Releasing end point's BLE connection back to application.
E: 118307 [DL]OnSrpClientNotification: Timed out waiting on server response
I: 119722 [DL]BLE GAP connection terminated (reason 0x13)
I: 119727 [DL]Current number of connections: 0/1
D: 119732 [DL]In DriveBLEState
E: 123542 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 132415 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 147476 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 173043 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 216478 [DL]OnSrpClientNotification: Timed out waiting on server response

However first I verified master with sha: 4bfdd42315e094d3dcd6bcc80f0ec5ec1f65015c and it was working, but then I noticed that my OTBR image is not up to date. I pulled the latest otbr image and after that it stopped working, so maybe that's the track where something could change. I'm going to investigate it, but I can't say nothing more about the reason on that moment.

kkasperczyk-no, 2021-05-13:
The commit that introduces regression for me is: https://github.com/openthread/ot-br-posix/commit/6f7fd7c21d2815dfc80bf7018a920cd0ebcf8551. Previous version works fine.

jmartinez-silabs, 2021-05-13:
Ok, I will try with the previous ot-br-possix version, and give an update here.

kkasperczyk-no, 2021-05-14:
It looks like in the newest OTBR versions some significant changes regarding SRP were introduced (https://github.com/openthread/openthread/commit/a40b42e29932104e2960e9cebb6ebec95c67bc4c) and that could in our case break backward compatibility for some reason. I verified locally that using newest OpenThread with the latest OTBR solves the problem, so I guess we should just bump OpenThread revision in CHIP.

jmartinez-silabs, 2021-05-14:
The Openthread repo has been restructured for the code found in`examples/platform` They are now in a different github repo and those repo have openthread as submodule. So for us in chip this `third_party/openthread/repo/examples/platforms` changes.

There is a decent amount of rework to be done to update this.

abtink, 2021-05-14:
I added a comment https://github.com/openthread/openthread/pull/6501#issuecomment-841450524 in response to @kkasperczyk-no question to describe the change form PR https://github.com/openthread/openthread/pull/6501.

closed 2021-06-04 by jmartinez-silabs