project-chip/connectedhomeip #6733: SRP updates should use new openthread DeleteService+AddService pattern

msandstedt, 2021-05-12:
#### Problem
Currently, SRP updates must call RemoveService() / AddService().  This was introduced with https://github.com/project-chip/connectedhomeip/pull/6539.

It was noted in this PR that on the server side, in place updates are possible.  However, from the client side, openthread disallows this because it is impermissible to modify the backing SRP advertisement data while a service is being actively advertised.

A solution to this problem is being introduced here: https://github.com/openthread/openthread/pull/6598.

This provides a DeleteService interface that deletes the service locally, but does not remove at the server side.  This provides an ability from the SRP client side to Delete locally, and then send the Add service request to perform in-place update.  Moving to this pattern would be simpler, more performant and would prevent any possibility of an attacker commandeering a service registration.  There are timers that typically prevent this, but it's still theoretically possible for a registrant to Remove, wait a very long time, malicious registration to occur, and ultimately for Add (which ought to have been an in-place update) to fail.

This was discussed here: https://github.com/openthread/openthread/issues/6586.  That issue was closed because, among other reasons, `OPENTHREAD_CONFIG_SRP_CLIENT_UPDATE_TX_DELAY` does mean the security problem is theoretical and / or unlikely to be easily exploitable.

Still, DeleteService+AddService is again simpler and more directly achieves the in-place update on the server side that we're looking for.

msandstedt, 2021-05-12:
@kkasperczyk-no

jwhui, 2021-06-02:
FYI - https://github.com/openthread/openthread/pull/6598 has been merged.

closed 2021-06-17 by woody-apple