project-chip/connectedhomeip #6746: [ZAP]: Need to support 'struct' construct for attribute in cluster xml definition file

yufengwangca, 2021-05-12:
#### Problem
Right now, we can not specify struct type in attribute definition.

 #### Proposed Solution
Need to support 'struct' construct for attribute in cluster xml definition file

tecimovic, 2021-05-16:
https://github.com/project-chip/zap/issues/151

tecimovic, 2021-05-16:
I've moved all the "chip_zap_support" labeled bugs into the zap repo itself under https://github.com/project-chip/zap/issues
I propose that we delete the `chip_zap_support` label, and all the bugs in this repo, and in the future simply open the issues in the zap repo itself, since it's much easier to keep track.

But given that I'm not sure how this repo is tracking "dependent issues" it's not my call.

closed 2021-06-10 by yufengwangca