project-chip/connectedhomeip #6749: [ZAP]: Allow specify the the length constraint of a list

yufengwangca, 2021-05-12:
#### Problem
Now we can specify 'list' type using "ARRAY" in a Attribute or Command definition:

<attribute side="server" code="0x0000" define="GROUPS" type="ARRAY" entryType="GroupState" length="254" writable="false" optional="false">groups</attribute>

But the length here is the maximum allowed bytes of each entryType. We need to have a way to specify the maximum element of the list defined by the CHIP spec 

 #### Proposed Solution
Allow specify the the length constraint of a list

tecimovic, 2021-05-16:
https://github.com/project-chip/zap/issues/151

tecimovic, 2021-05-16:
I've moved all the "chip_zap_support" labeled bugs into the zap repo itself under https://github.com/project-chip/zap/issues
I propose that we delete the `chip_zap_support` label, and all the bugs in this repo, and in the future simply open the issues in the zap repo itself, since it's much easier to keep track.

But given that I'm not sure how this repo is tracking "dependent issues" it's not my call.