project-chip/connectedhomeip #6753: [ZAP] ZAP is missing the construct to specify Event in its xml definition

yufengwangca, 2021-05-12:
#### Problem
Currently, we can specify 'attribute' and 'command' in ZAP cluster xml definition file, but not 'event'

 #### Proposed Solution
Need to support to specify event defined in CHIP spec in ZAP xml file.

tecimovic, 2021-05-16:
https://github.com/project-chip/zap/issues/152

tecimovic, 2021-05-16:
I've moved all the "chip_zap_support" labeled bugs into the zap repo itself under https://github.com/project-chip/zap/issues
I propose that we delete the `chip_zap_support` label, and all the bugs in this repo, and in the future simply open the issues in the zap repo itself, since it's much easier to keep track.

But given that I'm not sure how this repo is tracking "dependent issues" it's not my call.

closed 2021-06-10 by yufengwangca