project-chip/connectedhomeip #6755: [ZAP] Support list type with in struct type

yufengwangca, 2021-05-12:
#### Problem
Now we can use type 'ARRAY' to specify list type within 'attribute' or 'command', but not within user defined struct type.

 #### Proposed Solution
Support list type with in struct type

tecimovic, 2021-05-16:
https://github.com/project-chip/zap/issues/151

tecimovic, 2021-05-16:
I've moved all the "chip_zap_support" labeled bugs into the zap repo itself under https://github.com/project-chip/zap/issues
I propose that we delete the `chip_zap_support` label, and all the bugs in this repo, and in the future simply open the issues in the zap repo itself, since it's much easier to keep track.

But given that I'm not sure how this repo is tracking "dependent issues" it's not my call.

closed 2021-06-10 by yufengwangca