project-chip/connectedhomeip #6756: [ZAP] We need an XML reflection of feature bits

yufengwangca, 2021-05-12:
#### Problem
Currently, we can only specify Mandatory or Optional for an attribute or command definition, but a lot of them are controlled by its Feature bits in CHIP spec. Such as 'ERRCNT'. Vendor should be able to enable or disable those cluster element by enabling/disabling those Feature bits from UI.

 #### Proposed Solution
 We need an XML reflection of feature bits (

tecimovic, 2021-05-16:
Moved to zap: https://github.com/project-chip/zap/issues/150

tecimovic, 2021-05-16:
I've moved all the "chip_zap_support" labeled bugs into the zap repo itself under https://github.com/project-chip/zap/issues
I propose that we delete the `chip_zap_support` label, and all the bugs in this repo, and in the future simply open the issues in the zap repo itself, since it's much easier to keep track.

But given that I'm not sure how this repo is tracking "dependent issues" it's not my call.

closed 2021-06-10 by yufengwangca