project-chip/connectedhomeip #6774: Unable to compile ESP32 examples with CMake with latest CHIP master

dhrishi, 2021-05-13:
#### Problem
Unable to compile all-clusters-app, temperature-measurement-app and lock-app for ESP32 using CMake. The same work fine using automake. 
This seems to be a regression after the recent changes in the interaction model

erjiaqing, 2021-05-13:
This function is included in `libmain.a` for esp32, and the caller lives in `libCHIP.a`,

When look into the linker command line, I found `libmain,.a` sits before `libCHIP.a`, and is not wrapped by `start group` and `end group`

```
esp-idf/screen-framework/libscreen-framework.a
esp-idf/main/libmain.a
...
...
-Wl,--start-group
esp-idf/chip/lib/libCHIP.a
esp-idf/bt/libbt.a
-lbtdm_app
esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a
esp-idf/esp32_mbedtls/libesp32_mbedtls.a
-Wl,--end-group
...
```

In this case, some missing symbols in `libCHIP` cannot be filled by `libmain`, thus it caused failure.

I tried to add these two lines to config/esp32/components/chip/CMakeLists.txt , it can build

```
idf_component_get_property(main_lib main COMPONENT_LIB)
list(APPEND chip_libraries $<TARGET_FILE:${main_lib}>)
```

This can be a workaround but I think the final solution may be add similar code in CMakeLists.txt for libmain,.

cahty82008, 2021-05-14:
@erjiaqing  hi, i have the same issue, unfortunately add this two lines in config/esp32/components/chip/CMakeLists.txt doesn't work for me, could you pls have a check? here is my log:
```
root@ubuntu:/home/jsh/work_src/connectedhomeip/examples/all-clusters-app/esp32# idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/jsh/work_src/connectedhomeip/examples/all-clusters-app/esp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/jsh/work_src/connectedhomeip/examples/all-clusters-app/esp32"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /root/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /root/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /root/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /root/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /root/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/jsh/work_src/connectedhomeip/examples/all-clusters-app/esp32/sdkconfig
warning: the int symbol SPIFFS_BASE_ADDR (defined at /home/jsh/work_src/connectedhomeip/examples/common/m5stack-tft/repo/components/spiffs/Kconfig:3) has a non-int default 0x180000 (undefined)
Loading defaults file /home/jsh/work_src/connectedhomeip/examples/all-clusters-app/esp32/sdkconfig.defaults...
-- Found PythonInterp: /root/.espressif/python_env/idf4.2_py3.8_env/bin/python (found version "3.8.5") 
-- Found Perl: /usr/bin/perl (found version "5.30.0") 
-- App "chip-all-clusters-app" version: test_event_2_2012_04_22-165-g81
-- Adding linker script /home/jsh/tools/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/jsh/tools/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/jsh/tools/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/jsh/tools/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/jsh/tools/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/jsh/tools/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/jsh/work_src/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/jsh/tools/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/jsh/tools/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: QRCode app_trace app_update asio bootloader bootloader_support bt cbor chip coap console cxx driver efuse esp-tls esp32 esp32_mbedtls esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_ipc esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mkspiffs mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread screen-framework sdmmc soc spi_flash spidriver spiffs spiffs_image tcp_transport tcpip_adapter tft tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/jsh/work_src/connectedhomeip/examples/common/QRCode /home/jsh/tools/esp-idf/components/app_trace /home/jsh/tools/esp-idf/components/app_update /home/jsh/tools/esp-idf/components/asio /home/jsh/tools/esp-idf/components/bootloader /home/jsh/tools/esp-idf/components/bootloader_support /home/jsh/tools/esp-idf/components/bt /home/jsh/tools/esp-idf/components/cbor /home/jsh/work_src/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/config/esp32/components/chip /home/jsh/tools/esp-idf/components/coap /home/jsh/tools/esp-idf/components/console /home/jsh/tools/esp-idf/components/cxx /home/jsh/tools/esp-idf/components/driver /home/jsh/tools/esp-idf/components/efuse /home/jsh/tools/esp-idf/components/esp-tls /home/jsh/tools/esp-idf/components/esp32 /home/jsh/work_src/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/config/esp32/components/esp32_mbedtls /home/jsh/tools/esp-idf/components/esp_adc_cal /home/jsh/tools/esp-idf/components/esp_common /home/jsh/tools/esp-idf/components/esp_eth /home/jsh/tools/esp-idf/components/esp_event /home/jsh/tools/esp-idf/components/esp_gdbstub /home/jsh/tools/esp-idf/components/esp_hid /home/jsh/tools/esp-idf/components/esp_http_client /home/jsh/tools/esp-idf/components/esp_http_server /home/jsh/tools/esp-idf/components/esp_https_ota /home/jsh/tools/esp-idf/componen [... cut ...] taElement(chip::app::CommandDataElement::Parser&)':
/home/jsh/work_src/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/app/CommandSender.cpp:157: undefined reference to `chip::app::DispatchSingleClusterCommand(unsigned short, unsigned char, unsigned char, chip::TLV::TLVReader&, chip::app::Command*)'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

erjiaqing, 2021-05-14:
> @erjiaqing hi, i have the same issue, unfortunately add this two lines in config/esp32/components/chip/CMakeLists.txt doesn't work for me, could you pls have a check? here is my log:
>
> ``` [...]

You need to add the two lines above `target_link_libraries`

i.e. The file should looks like:

```
...
...
idf_component_get_property(main_lib main COMPONENT_LIB)
list(APPEND chip_libraries $<TARGET_FILE:${main_lib}>)

target_link_libraries(${COMPONENT_LIB} INTERFACE -Wl,--start-group 
                                                ${chip_libraries}
                                                $<TARGET_FILE:mbedcrypto> $<TARGET_FILE:${esp32_mbedtls_lib}> 
                                                -Wl,--end-group)

# Make the component dependent on our CHIP build
add_dependencies(${COMPONENT_LIB} chip_gn)
```

We may find a proper (and clean) way to fix this issue.

erjiaqing, 2021-05-14:
@dhrishi 

If this issue is blocking, can we use the workaround first and find the clean fix to this issue later? Since `libmain` is the default component for esp32 (correct?), it should work for most cases.

cahty82008, 2021-05-14:
@erjiaqing  thanks a lot, it's working now

dhrishi, 2021-05-15:
>If this issue is blocking, can we use the workaround first and find the clean fix to this issue later? Since libmain is the default component for esp32 (correct?), it should work for most cases.

@erjiaqing  Yes, have raised a PR: #6849

PR #6849 (2021-05-17): ESP32: Fix compilation of examples with CMake
 #### Problem
ESP32 examples fail with CMake.

 #### Summary of Changes
As described here: https://github.com/project-chip/connectedhomeip/issues/6774#issuecomment-840447955

Fixes #6774
closed 2021-05-17 by andy31415