project-chip/connectedhomeip #6775: build issue while generating efr32 light app on raspberry pi with Ubuntu 20.10

somshekharjuja, 2021-05-13:
#### Problem
<what's wrong or missing, please include any applicable:
I am trying to build efr32 light app in my raspberry pi which has Ubuntu 20.10 runnning  and gcc version is 10.2 . I see that , i am getting build error and error logs are pasted below.
 * expected behavior
  There should not be any build error
 * actual behavior
  I am getting build error while generating efr32 light app .
 * steps to reproduce
  source script/activate.sh
export EFR32_BOARD=BRD4161A
gn gen out/debug
ninja -C out/debug
  
 * system configuration
  ubuntu 20.10 running in Raspberry Pi 4

 * screenshots, images, logs, etc.
 >
connectedhomeip/src/credentials/libCredentials.GenerateChipX509Cert.cpp.o
../../third_party/connectedhomeip/src/credentials/GenerateChipX509Cert.cpp: In function 'CHIP_ERROR chip::Credentials::{anonymous}::EncodeChipDNs(chip::Credentials::{anonymous}::ChipDNParams*, uint8_t, chip::ASN1::ASN1Writer&)':
../../third_party/connectedhomeip/src/credentials/GenerateChipX509Cert.cpp:280:92: error: expected ')' before 'PRIX64'
  280 |                 snprintf(reinterpret_cast<char *>(chipAttrStr), sizeof(chipAttrStr), "%016" PRIX64, params[i].Value);
      |                         ~                                                                  ^~~~~~~
      |                                                                                            )
../../third_party/connectedhomeip/src/credentials/GenerateChipX509Cert.cpp:280:90: error: conversion lacks type at end of format [-Werror=format=]
  280 |                 snprintf(reinterpret_cast<char *>(chipAttrStr), sizeof(chipAttrStr), "%016" PRIX64, params[i].Value);
      |                                                                                          ^
../../third_party/connectedhomeip/src/credentials/GenerateChipX509Cert.cpp:280:86: error: too many arguments for format [-Werror=format-extra-args]
  280 |                 snprintf(reinterpret_cast<char *>(chipAttrStr), sizeof(chipAttrStr), "%016" PRIX64, params[i].Value);
      |                                                                                      ^~~~~~
At global scope:
cc1plus: error: unrecognized command line option '-Wno-unknown-warning-option' [-Werror]
[image]

 #### Proposed Solution
<suggested fix, suggested enhancement>

somshekharjuja, 2021-05-26:
Can anyone please help me with this issue

woody-apple, 2022-01-25:
Sorry about that - we're in progress of reviewing older issues. Is this still an issue given how old the SDK this was tested against? Please re-open if so. Thanks!

closed 2022-01-25 by woody-apple