project-chip/connectedhomeip #6805: Switch NetworkInterfaceType Name member to CHAR_STRING

bzbarsky-apple, 2021-05-14:
#### Problem
We don't support CHAR_STRING in structs yet.  So for now the Name of NetworkInterfaceType in `src/app/zap-templates/zcl/custom-types.xml` is OCTET_STRING.

 #### Proposed Solution
Once CHAR_STRING is supported in structs, use it here.

@vivien-apple

bzbarsky-apple, 2021-05-14:
Might be a dup of https://github.com/project-chip/connectedhomeip/issues/6112

mrjerryjohns, 2022-01-31:
I can't seem to find `NetworkInterfaceType` in the cluster specs anymore. More-over, char string is now supported. Marking as stale.

closed 2022-01-31 by mrjerryjohns

bzbarsky-apple, 2022-02-01:
> I can't seem to find NetworkInterfaceType in the cluster specs anymore

In the spec it's called "NetworkInterface struct".  In our XML it's NetworkInterfaceType.

That said, its Name is CHAR_STRING in the XML, so we're good.