project-chip/connectedhomeip #6811: Do not use /tmp directory for persistent storage

bluebin14, 2021-05-14:
#### Problem
chip-tool uses /tmp as a directory for persistent storage, see examples/chip-tool/config/PersistentStorage.cpp.

This is not good since by definition the contents of /tmp are not persistent.

 #### Proposed Solution
Use anything else but not /tmp.

yunhanw-google, 2021-06-02:
@bzbarsky-apple @yufengwangca any reasoning for us to use tmp?

bzbarsky-apple, 2021-06-02:
@vivien-apple might know, though he might also be off for the rest of the week....

In any case, we could try putting this somewhere else; the question is where.  `${HOME}/.matter` or something?

tcarmelveilleux, 2021-07-23:
I would argue that the location of the settings should never be assumed to have a default, and a command-line argument should have the path for the storage root, on hosts. The test scripts that run the different tools should pass-in appropriate values. The reason to avoid a default is that defaults just work if a single "thing" is used, or if we make strong assumption about how the storage is done, such that collisions of stored state is not possible.

mspang, 2021-07-26:
/tmp honestly sounds right to me since we just base64 encoding C structs...

Without adopting a backwards compatible storage format, long term persistence is impossible anyway.

mspang, 2021-07-26:
(the current format is not even ABI stable so it's really quite problematic)

kpschoedel, 2021-07-27:
Any hard-coded location is painful when using multiple devices.

bluebin14, 2021-09-18:
Based on input a more flexible solution is needed that also allows multi controller and multi device support. Basically adding some command line options. Closing this and opening https://github.com/project-chip/connectedhomeip/issues/9818

closed 2021-09-18 by bluebin14