project-chip/connectedhomeip #6828: chip-device-ctrl, see Ble endpoint pool FULL issue in tot

yunhanw-google, 2021-05-14:
#### Problem
In mac, run chip-device-ctrl, see Ble endpoint pool FULL issue

chip-device-ctrl > connect -ble 2976 34567890 1
Device is assigned with nodeid = 1
[1621017719114] [0x196d03] CHIP: [DL] Chip stack locking error at '../../src/system/SystemLayer.cpp:204'. Code is unsafe/racy
[1621017719114] [0x196d03] CHIP: [BLE] NewConnection
[1621017719116] [0x196d03] CHIP: [BLE] Assigned local session key ID 0
[1621017719116] [0x196d03] CHIP: [DL] Chip stack locking error at '../../src/system/SystemLayer.cpp:204'. Code is unsafe/racy
[1621017719116] [0x196d03] CHIP: [IN] Message appended to BLE send queue
[1621017719116] [0x196d03] CHIP: [BLE] Sent PBKDF param request
[1621017719116] [0x196d03] CHIP: [CTL] SyncSetKeyValue on StartKeyID
[1621017719169] [0x196d1e] CHIP: [BLE] CBManagerState: ON
[1621017735129] [0x197012] CHIP: [BLE] Connecting to device with discriminator: 2976
[1621017739497] [0x196d1e] CHIP: [IN] BleConnectionComplete: endPoint 0x10f49b068
[1621017739497] [0x196d1e] CHIP: [DL] Chip stack locking error at '../../src/system/SystemLayer.cpp:204'. Code is unsafe/racy
[1621017739497] [0x196d1e] CHIP: [BLE] Ble endpoint pool FULL
[1621017739497] [0x196d1e] CHIP: [IN] Clearing BLE pending packets.
[1621017739497] [0x196d1e] CHIP: [IN] BleConnection Error: Ble Error 6002 (0x00001772): No more BLE endpoints
[1621017740350] [0x196d1e] CHIP: [BLE] subscribe complete, ep = 0x10f49b068
[1621017740396] [0x197012] CHIP: [BLE] peripheral chose BTP version 3; central expected between 2 and 3
[1621017740396] [0x197012] CHIP: [BLE] using BTP fragment sizes rx 20 / tx 128.
[1621017740396] [0x197012] CHIP: [BLE] local and remote recv window size = 3
[1621017740396] [0x197012] CHIP: [DL] Chip stack locking error at '../../src/system/SystemLayer.cpp:204'. Code is unsafe/racy
[1621017740396] [0x197012] CHIP: [IN] BLE EndPoint 0x10f49b068 Connection Complete
[1621017745436] [0x197012] CHIP: [DL] Chip stack locking error at '../../src/system/SystemLayer.cpp:204'. Code is unsafe/racy
[1621017749118] [0x196d35] CHIP: [CTL] SyncSetKeyValue on StartKeyID
Failed to establish secure session to device: 4050
[1621017749118] [0x196d35] CHIP: [BLE] PASESession timed out while waiting for a response from the peer. Expected message type was 33
[1621017749118] [0x196d35] CHIP: [CTL] SyncSetKeyValue on StartKeyID
CHIP Error 4050 (0x00000FD2): Timeout
Failed to establish secure session to device: 4050
chip-device-ctrl > [1621017750475] [0x196d1e] CHIP: [DL] Chip stack locking error at '../../src/system/SystemLayer.cpp:204'. Code is unsafe/racy
[1621017755516] [0x196d1e] CHIP: [DL] Chip stack locking error at '../../src/system/SystemLayer.cpp:204'. Code is unsafe/racy
[1621017760555] [0x197012] CHIP: [DL] Chip stack locking error at '../../src/system/SystemLayer.cpp:204'. Code is unsafe/racy

yunhanw-google, 2021-05-14:
@vivien-apple could you take a look at this issue?

yunhanw-google, 2021-05-15:
@bzbarsky-apple @woody-apple  need your team to have a look at this issue, impacting TE3.

vivien-apple, 2021-05-19:
Duplicate of #6863

closed 2021-05-26 by yunhanw-google